project-chip/connectedhomeip #10129: Invalid Access to ExchangeContext and Session values after Expiry during Error Propagation

samadDotDev, 2021-09-30:
During the write interaction, If the session expires leading to closing the associated exchanges, the session value is cleared before the exchange context is removed. 

```cpp
void ExchangeContext::OnConnectionExpired()
{
    // Reset our mSecureSession to a default-initialized (hence not matching any
    // connection state) value, because it's still referencing the now-expired
    // connection.  This will mean that no more messages can be sent via this
    // exchange, which seems fine given the semantics of connection expiration.
    mSecureSession.ClearValue();
```

However, this can potentially lead to crash during WriteError which uses this exchange context and session value out of it to determine the nodeId in propagating the failure callback. For instance, `DeviceControllerInteractionModelDelegate::WriteResponseError` proceeds to the following
```cpp
bool IMWriteResponseCallback(const chip::app::WriteClient * writeClient, EmberAfStatus status)
{
    ChipLogProgress(Zcl, "WriteResponse:");
    LogStatus(status);

    Callback::Cancelable * onSuccessCallback = nullptr;
    Callback::Cancelable * onFailureCallback = nullptr;
    NodeId sourceNodeId                      = writeClient->GetSourceNodeId();
...
```
Where the WriteClient will attempt to retrieve the nodeId using the evicted session handle:
```cpp
    NodeId GetSourceNodeId() const
    {
        return mpExchangeCtx != nullptr ? mpExchangeCtx->GetSecureSession().GetPeerNodeId() : kUndefinedNodeId;
    }
```
It's similar to Issue #10047 but occurs during Write or Other Interactions that access the values from evicted exchangeContext or sessionHandle. It can be reproduced by calling `SecureSessionMgr::ExpirePairing` during such interaction.

msandstedt, 2021-10-01:
CC @bzbarsky-apple , @yufengwangca

bzbarsky-apple, 2021-10-01:
Seems like we have two sets of information inside `mSecureSession`: metadata about the session (e.g. node ids) that is still valid even if the session is gone and session data (e.g. key ids) that is not.  And maybe a third bit, which is references to the connection.

Presumably `OnConnectionExpired` should clear the connection references, and maybe the "no longer valid once session is gone" state, but not the "what was this session about?" state.

And we need tests.

@pan-apple

yufengwangca, 2022-02-02:
need to check if this issue still exists

woody-apple, 2022-02-08:
@yufengwangca Please re-open if this still happens.

closed 2022-02-08 by woody-apple