project-chip/connectedhomeip #10161: [session] [crypto] Sessions should use correct Source Node ID in NONCE.

turon, 2021-10-02:
#### Problem
Based on current testing of chip-tool with all-clusters-app, it appears all nodes are using NodeID = 0 for the nonce/IV.
This is perhaps not the end of the world security-wise, as sessions will typically have a unique key.
Per spec: "For unicast secure session communication, both the Source Node ID and Destination Node ID SHOULD be omitted, as they are derivable from the session context referenced by the Session ID."
NOTE: utilizing the proper Source Node ID for the nonce will complicate the sniffer dissector and other such tools.

#### Proposed Solution
Nodes need to encrypt with the source NodeID they advertised/negotiated during session establishment.
SessionManager and Session objects need to track the peer's NodeID and use that address for the nonce in decryption.

Specifically, `CryptoContext::GetIV` should not call `header.GetSourceNodeId()` unless SessionType = group and Sflag=1.
Instead, the peer node id as authenticated during CASE should be used.

For PASE, this is less of an issue.

holbrookt, 2022-02-01:
@turon is this needed for 1.0?

woody-apple, 2022-02-08:
Per spec, this is a should, so this can wait until v1.X as long as we don't have a security issue here.

tcarmelveilleux, 2022-02-24:
It was intended in the security design for CASE that the source node ID would *always* be included in the nonce, but from the secure session rather than the header. The SHOULD mentioned above is actually for something else.

turon, 2022-05-24:
Fixed #16098

closed 2022-05-24 by turon