project-chip/connectedhomeip #10187: TLV tag use in SetupPayload does not seem to match spec

bzbarsky-apple, 2021-10-04:
#### Problem
In SetupPayload.cpp we have:
```
bool IsCHIPTag(uint8_t tag)
{
    return tag >= (1 << kRawVendorTagLengthInBits);
}

bool IsVendorTag(uint8_t tag)
{
   return tag < (1 << kRawVendorTagLengthInBits);
}
```

but in the spec "5.1.4.1. Manufacturer-specific Elements" says:

> Manufacturer-specific elements SHALL be tagged with context-specific tags that have semantics which are defined by the vendor for use in the products using their Vendor ID, and SHALL use tag values 0x80 to 0xFF.  Tag values 0x00 to 0x7F are reserved to indicate CHIP-common elements.
    
which seems exactly backwards from the code.

What exactly is going on here?  @shana-apple 

 #### Proposed Solution
Sort out what's going on, and fix or document to make it clear how this code and the spec interact.  Add tests as needed.

tcarmelveilleux, 2021-10-04:
Suggest:

bool IsCommonTag(uint8_t tag)
{
    return tag < 0x80;
}

bool IsVendorTag(uint8_t tag)
{
   return tag >= 0x80;
}
```

leorozendaal, 2021-10-05:
@bzbarsky-apple  in initial spec the range of CHIP/Matter vs vendor was opposite and was changed . Maybe the code you found predates that change

woody-apple, 2022-02-08:
Marked as spec.

PR #17122 (2022-04-07): Fix TLV tags in the Onboarding payload and update unit tests
#### Problem
Fixes #10187 

#### Change overview
Update the Manufacture-specific and CHIPCommon optional tags in the Onboarding Payload. Updated unit tests to reflect this change.

#### Testing
- Updated Unit tests
- Sanity-check: Ran OTA test to ensure commissioning and OTA transfer is unaffected
closed 2022-04-07 by andy31415