project-chip/connectedhomeip #10190: CASE resumption fails with openssl due to null pointers being used

bluebin14, 2021-10-04:
#### Problem
<what's wrong or missing, please include any applicable:
 * CASE resumption fails with openssl due to null pointers being used. The failure is in CASESession::GenerateSigmaResumeMIC at calling AES_CCM_encrypt with null pointers.
 >

 #### Proposed Solution
Use non-zero pointers when calling AES_CCM_encrypt or fix the AES_CCM_encrypt implementation for openssl

bluebin14, 2021-10-04:
@pan-apple

bluebin14, 2021-10-05:
#10198 fixes the issue for  the null source case but not for the null destination case. CSession::GenerateSigmaResumeMIC calls AES_CCM_encrypt with both plaintext and ciphertext being nullptr and still fails.

pan-apple, 2021-10-05:
@bluebin14 , the unit tests are testing that the destination can be `nullptr`. Are you still seeing the issue when you use it?

bluebin14, 2021-10-05:
The unit tests were still using non null destinations.

PR #10198 (2021-10-04): Enable use of nullptr for 0 length plaintext/ciphertext in crypto
#### Problem
* Fixes #10190

#### Change overview
OpenSSL APIs do not support using `nullptr` for 0 length plaintext for encrypt, and 0 length cipher-text for decrypt operations. Expectation is to use an empty string instead of nullptr.

This PR uses a temporary empty string if the user has provided a nullptr and 0 length for the corresponding value. It also adds more checks and tests to ensure that API behaves as expected.
 

#### Testing
Added more unit tests to `CHIPCryptoPALTest` suite.