project-chip/connectedhomeip #10192: resolve fails with minimal mdns enabled on test_event_6 branch

pramodhg123, 2021-10-04:
While building python contoller with default value of mdns as minimal the resolve is failing
branch: test+event 6 commit id:    96003a703e34b4a663f45e66e4f3de92e5279e66
resolve <conpressed fabric id> <nodeid> should work for minimal mdns as well

chip-device-ctrl > resolve 7993271286770236299 1
[1633343626.774423][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343626.774831][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343626.775171][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343626.775579][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343627.775242][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343627.775555][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343627.775951][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343627.776307][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343629.777623][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343629.777878][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343629.778183][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343629.778457][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343633.780745][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343633.781056][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343633.781522][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343633.781885][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343641.782493][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343641.782745][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343641.783110][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343641.783380][43783:43791] CHIP:DIS: mDNS broadcast success
[1633343657.784759][43783:43791] CHIP:DIS: Timeout waiting for mDNS resolution.
[nrf_ctrl_oct4.txt](https://github.com/project-chip/connectedhomeip/files/7279982/nrf_ctrl_oct4.txt)

andy31415, 2021-10-04:
Could you provide details to reproduce?

What HW (assume NRF board?) and what did you run where? 

Did provisioning succeed before resolution?

andy31415, 2021-10-04:
Since our timezone does not overlap well, some extra info could help:

- how was chip-device-ctrl compiled?
- is avahi running on the device that runs chip-device-ctrl?
- can avahi discover the given node at all?
- if capturing mDNS traffic (port 5353 udp) are the broadcasts visible and does the nrf device reply to them? 

Trying to narrow the issue down to something like "conflict with system avahi" or "failure to properly broadcast" or "failure to interpret broadcastt responses"

andy31415, 2021-10-06:
Steps I received for reproduction:

1. Use steps in https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/nrfconnect to build the lighting app

2. Ensure OTBR is installed (docker/standalone)

3. Python controller built using `./scripts/build_python.sh --chip_detail_logging true` on the RPi

4. Follow log file for commands (using `./chip-device-ctrl`)

```
connect -ble 3840 20202021 1

zcl NetworkCommissioning AddThreadNetwork 1 0 0 operationalDataset=000300000f02081111111122222222051000112233445566778899aabbccddeeff01021234 breadcrumb=0 timeoutMs=1000

zcl NetworkCommissioning EnableNetwork 1 0 0 networkID=hex:1111111122222222 breadcrumb=0 timeoutMs=1000

close-ble

resolve 7993271286770236299 1
```

andy31415, 2021-10-06:
Operational dataset can be read from the OTBR using `sudo ot-ctl dataset active -x`

andy31415, 2021-10-06:
https://github.com/openthread/ot-br-posix/commit/a4880eb0a782adb9e6104cb00aa285d0be3a669a was reported by @Damian-Nordic as a desirable fix for OTBR

andy31415, 2021-10-06:
Linux TE6 branch worked fine for both provisioning and resolution:

```
chip-device-ctrl > resolve 12211974628394890774 178945
[1633551216.959959][15577:15584] CHIP:DIS: Attempt to mDNS broadcast failed:  ../../src/inet/IPEndPointBasis.cpp:902: OS Error 0x02000065: Network is unreachable
[1633551216.960035][15577:15584] CHIP:DIS: Attempt to mDNS broadcast failed:  ../../src/inet/IPEndPointBasis.cpp:902: OS Error 0x02000063: Cannot assign requested address
[1633551216.960265][15577:15584] CHIP:DIS: mDNS broadcast success
[1633551216.960423][15577:15584] CHIP:DIS: mDNS broadcast success
[1633551216.960562][15577:15584] CHIP:DIS: mDNS broadcast success
[1633551216.960653][15577:15584] CHIP:DIS: Attempt to mDNS broadcast failed:  ../../src/inet/IPEndPointBasis.cpp:902: OS Error 0x02000065: Network is unreachable
[1633551216.960796][15577:15584] CHIP:DIS: mDNS broadcast success
[1633551216.960906][15577:15584] CHIP:DIS: mDNS broadcast success
[1633551217.010761][15577:15584] CHIP:DIS: Node ID resolved for 0x000000000002BB01 to [fd3d:e88:d817:1d1f:33a9:cc8f:2fa0:7794]:5540
Node address has been updated
```

andy31415, 2021-10-07:
The 'Expected message type 49' is 'CASE_SigmaR2  = 0x31,'