project-chip/connectedhomeip #10221: BLE Advertising device name is undesirable

jimlyall-q, 2021-10-05:
#### Problem
A name is not part of to-spec Matter advertising. If you add it, put it in the Scan Response instead. Spec text:
In order to reduce 2.4 GHz spectrum congestion due to active BLE scanning, and to extend battery life in battery-powered devices, all critical data used for device discovery is contained in the Advertising Data rather than the Scan Response Data. This allows a BLE Commissioner to passively scan (i.e., not issue Scan Requests upon receiving scannable advertisements) and still be able to receive all information needed to commission a device.

 * Matter device name should not appear in BLE advertising
 * Matter device name appears in BLE advertising for serveral vendors
 * steps to reproduce - passively scan devices with nrf-connect app

 #### Proposed Solution
Note that if additional vendor-specific information is to be conveyed and does not fit within the Advertising Data, it may be included in the Scan Response Data. See <> for details on including vendor-specific information.

turon, 2022-02-01:
BLE advertisement must be spec compliant.  
Removing tag `v1_triage_split_4`.

woody-apple, 2022-05-25:
SDK Spec Review: We need to make sure there's a test case for this. Assigning to @cjandhyala for now to follow up to create a cert test.

cjandhyala, 2022-05-25:
Followed-up with an issue in the test plan, https://github.com/CHIP-Specifications/chip-test-plans/issues/1355 , once the test case is added, QA will run this test as part of certification test

andy31415, 2022-06-15:
Is advertising the name undesirable (sure is) or also wrong as per spec?

I do not see SHALL/SHOULD language regarding this in the spec. The text is:

```
In order to reduce 2.4 GHz spectrum congestion due to active BLE scanning, and to extend battery life in battery-powered devices, all critical data used for device discovery is contained in the Advertising Data rather than the Scan Response Data.
```

So we say "all that is needed is contained in advertising data and we do not need to scan". I do not see "Matter device name should not appear in BLE advertising" in the spec. @cjandhyala was the test plan added? Should it be adjusted?

andy31415, 2022-06-15:
I believe cert should validate: `Advertising data contains all critical data for device discovery` (need a list of what this is though ....). It should *NOT* check for `advertising data does not contain device name` as the spec does not explicitly mandate that.

turon, 2022-06-15:
I agree.  Certification requires the specified payload exists.  If a vendor wants to add extra TLVs such as device name to their advertisement, that is fine.  Those extra TLVs may:

0) not exist at all (no device name);
1) fit in the ADV data;
2) fit in the SCAN response;
3) be partially encoded in ADV data and SCAN response (flow over)

All of the above are valid.  Certification must simply validate the Matter specified portions.

woody-apple, 2022-06-21:
Spec Issue Review: Unassigning @cjandhyala given he's OoO. @turon Can you do a quick pass to see how the SDK is here in relation to spec conformance?

closed 2022-07-04 by andy31415