project-chip/connectedhomeip #10251: Implement AccessControlProvider storage format

mlepage-google, 2021-10-05:
- design storage format
- implement load/save functionality
- ensure system is robust (data integrity etc.)
- lean and fast also important (in that order)
- defer extensions for now, but keep them in mind

mlepage-google, 2021-11-16:
PR #11550 adds an in-memory implementation that uses a single array of POD "storage" structs for the entry data. That format is already quite terse.

Therefore, what mainly needs to happen here:
- in AccessControl::Init, check for any stored ACL data, and if it exists, load it into the ACL storage area in-memory
- otherwise don't, and let the in-memory storage be the starting data
- in AccessControl::CreateEntry, write the new entry (the last one in the in-memory ACL storage) to flash storage
- in AccessControl::UpdateEntry, write the updated entry (just that one, could be anywhere in ACL) to flash storage
- in AccessControl::DeleteEntry, write the entries from the deleted index to the end of the used indexes, to flash storage (since they moved to fill the hole)

Probably wise to consider some kind of mini header for the flash storage, containing a version etc.

mlepage-google, 2022-01-27:
@harimau-qirex is putting the simple approach (down in access control module) in PR #14253 so we can have some persistence in the interim, then will refactor it out into the cluster with improved storage format and more use of generated code.

harimau-qirex, 2022-01-28:
The remaining work is for each platform to call chip::Access::Examples::SetAccessControlDelegateStorage (defined in src/access/examples/ExampleAccessControlDelegate.h) with a PersistentStorageDelegate implementation. Linux and MacOS call this in examples/platform/linux/AppMain.cpp with an implementation of PersistentStorageDelegate backed by the KVS manager.

The keys passed into the PersistentStorageDelegate are defined in src/lib/support/DefaultStorageKeyAllocator.h as AccessControlList and AccessControlEntry. By default there are 48 entries (numbered 0-47), but that is configurable by setting CHIP_CONFIG_MAX_DEVICE_ADMINS and CHIP_CONFIG_EXAMPLE_ACCESS_CONTROL_MAX_ENTRIES_PER_FABRIC.

PR #12814 (2021-12-21): Add persistent storage for the example ACL implementation.
#### Problem
The example AccessControlList needs to persist its data in order for the access control to function after a device has been turned off.
Fixes #10251 

#### Change overview
This adds a simple TLV-encoding for the data in the example AccessControl implementation, and stores this information in a persistent storage delegate. Each entry is stored separately using the hierarchical key format in DefaultStorageKeyAllocator, and the data is versioned using a single version number stored in a different key.

#### Testing
This was m [... cut]
PR #14680 (2022-02-04): Move the access control ownership and initialization to Server.
#### Problem
The simple ACL storage added in #14253 is only supported by Linux and MacOS. By moving the ownership of the AccessControl object to Server and using the platform storage delegate that it owns, all platforms that use Server will get ACL persistence.
Fixes #10251.

#### Change overview
* Move the ownership of the AccessControl object from the ACL cluster to Server.
* Remove the ACL storage delegate from the AppMain.cpp used by Linux and MacOS.
* Initializing ACL without storage or stored ACL data is no longer considered [... cut]
closed 2022-02-04 by bzbarsky-apple