project-chip/connectedhomeip #10261: enum prefix  duplicates the enum class name in cluster object

yunhanw-google, 2021-10-05:
Not really part of this PR, but why do we have that "DOOR_LOCK_OPERATION_EVENT_CODE_" prefix that just duplicates the enum class name?  Followup to drop all those, please.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/10159#discussion_r722637243_

bzbarsky-apple, 2021-10-06:
So the reason this is a problem is that some of the enum values have names that are not valid C++ identifiers because they start with a digit.  An example:
```
src/app/zap-templates/zcl/data-model/chip/onoff-cluster.xml:    <item name="20PercenterDimUpIn_0p5SecondsThenFadeToOffIn_1Second" value="0x00"/>
```
which becomes `20_PERCENTER_DIM_UP_IN_0P5_SECONDS_THEN_FADE_TO_OFF_IN_1_SECOND` when run through `asDelimitedMacro`  (as we do right now).

My proposal would be:
1. Use `asUpperCamelCase` instead of `asDelimitedMacro`, because it's less shouty.
2. `asUpperCamelCase` already strips out some characters that are not valid inside C++ identifiers.  Would it make sense to modify it to check for leading digit and prepend `'_'`?

@vivien-apple @tecimovic

bzbarsky-apple, 2022-01-29:
This was fixed by prefixing with `k` instead.

closed 2022-01-29 by bzbarsky-apple