project-chip/connectedhomeip #10332: InteractionModelEngine's CommandHandler pool is unsafely destroying handlers

mrjerryjohns, 2021-10-07:
#### Problem

In the IM engine:

```
    // Since modifying the pool during iteration is generally frowned upon,
    // I've chosen to just destroy the object but not necessarily de-allocate it.
    //
    // This poses a problem when shutting down and restarting the stack, since the
    // IMEngine is a statically constructed singleton, and this lingering state will
    // cause issues.
    //
    // This doesn't pose a problem right now because there shouldn't be any actual objects
    // left here due to the synchronous nature of command handling, but it should be fixed.

    mCommandHandlerObjs.ForEachActiveObject([](CommandHandler * obj) -> bool {
        obj->~CommandHandler();
        return true;
    });
```

woody-apple, 2022-02-08:
@mrjerryjohns Is this required for V1.0? Not clear if things work now, and could be better or we have an issue?

mrjerryjohns, 2022-06-23:
Has since been fixed.

closed 2022-06-23 by mrjerryjohns