project-chip/connectedhomeip #10337: Adding multiple admins fails with CASE error on TE6 branch

cjandhyala, 2021-10-07:
Adding 2nd admin(2nd python control) fails On TE6 branch (82bcd4e36f4856a4216cb76a16116faeec40e223).

Steps:
1. Launch lighting app
2. commission the lighting app with first python controller over BLE
3. Open commissioning window
4. add the 2nd admin (2nd python controller running on different raspi than 1st controller) using connect -IP cmd
5. 2nd commissioner resolves the node, but fails to establish CASE session. 

Attached logs, 
[1st_controller.txt](https://github.com/project-chip/connectedhomeip/files/7307342/1st_controller.txt)
[2ndController.txt](https://github.com/project-chip/connectedhomeip/files/7307343/2ndController.txt)
[accessory_log.txt](https://github.com/project-chip/connectedhomeip/files/7307344/accessory_log.txt)

bluebin14, 2021-10-08:
It works for me, however I used our own BLE/Thread device instead of Lighting app. It looks like an mdns/routing problem, can you try with both controllers on same raspi as a test?

bzbarsky-apple, 2021-10-08:
So it looks like controller2 correctly resolves the device.  It sends a Sigma1 (line 383 of 2ndController.txt).  The device receives it (line 2358 of accessory_log.txt) and sends a Sigma2 (line 2394).  The commissioner receives the Sigma2 but (starting line 384 in 2ndController.txt):
```
[1633646951.870298][3191:3199] CHIP:EM: Received message of type 0x31 with vendorId 0x0000 and protocolId 0x0000 on exchange 13378i
[1633646951.870406][3191:3199] CHIP:EM: Handling via exchange: 13378i, Delegate: 0x(nil)
[1633646951.870469][3191:3199] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:298: CHIP Error 0x0000002A: Invalid message type
```
So most pressingly, the exchange delegate is null, so the Sigma2 message is not delivered to the CASE state machine and then the handshake fails.

I am also not sure what the deal is with the "invalid message type" message there.  That means this check tested true:
```
        if (ec->IsEncryptionRequired() != packetHeader.GetFlags().Has(Header::FlagValues::kEncryptedMessage))
```
which is pretty odd in this case.

Anyway, after this the commissioner retransmits Sigma1 a few times, and the other side retransmits Sigma2 a few times, but they all get dropped as duplicates.  And in any case, on the server side we had, 5 seconds after sending Sigma2 (line 2396 in the accessory log):
```
[1633646956.576393][3488:3488] CHIP:SVR: Commissioning failed with error ../../examples/lighting-app/linux/third_party/connectedhomeip/src/platform/DeviceControlServer.cpp:48: CHIP Error 0x00000032: Timeout
[1633646956.576453][3488:3488] CHIP:SVR: Commissioning failed (attempt 1): ../../examples/lighting-app/linux/third_party/connectedhomeip/src/platform/DeviceControlServer.cpp:48: CHIP Error 0x00000032: Timeout
```
which happens _before_ the retransmits.

But the big question is what closed the CASE establishment exchange on 2ndController after sending Sigma1, so that the delegate ended up null.  @pan-apple any ideas?

msandstedt, 2021-12-31:
https://github.com/project-chip/connectedhomeip/issues/12821 will contribute to this.

bzbarsky-apple, 2022-06-27:
@cjandhyala Is this still an issue?

woody-apple, 2022-07-22:
Issue Review: Closing issues prior to TE8. Please re-file if this is still an issue on newer SDKs

closed 2022-07-22 by woody-apple