project-chip/connectedhomeip #10379: Broken API for Diagnostics cluster.

sweetymhaiske, 2021-10-11:
#### Problem
* Currently, Getter methods for some WiFi Diagnostic clusters attributes return 0 when unimplemented. However, specs demand NULL value.
Spec -
`A value of NULL SHALL indicates that the interface is not currently configured or operational.`

bzbarsky-apple, 2021-10-11:
@yufengwangca Do you have time to look at this?

@mrjerryjohns To implement this we need to decide how nullables will be represented for purposes of DataModel::Encode.  I still think we should treat them the way we treated List compared to Span and have a Nullable that derives from Optional....

mrjerryjohns, 2021-11-19:
@yufengwangca is this done?

bzbarsky-apple, 2021-11-20:
This is not done.  These attributes all need to get marked nullable in the XML and the spec-expected behavior around when null is returned needs to be implemented.

holbrookt, 2022-02-01:
This seems important but hasn't seen any activity since November. Does this need to be in for 1.0?

bzbarsky-apple, 2022-02-01:
Yes, unless the spec changes.  Right now we are not following the spec for this cluster on the server, and our clients can't parse spec-compliant server responses.

PR #14824 (2022-02-08): Align Generic Diagnostics cluster with the latest spec
#### Problem
What is being fixed?  Examples:
* Currently, Getter methods for some Generic Diagnostic clusters attributes return 0 on failure. However, specs demand NULL value.
* Fixes #10379 

#### Change overview
Align Generic Diagnostics cluster with the latest spec

#### Testing
How was this tested? (at least one bullet point required)
yufengw@yufengw-SEi:~/connectedhomeip/out/debug/standalone$ ./chip-tool generaldiagnostics read network-interfaces 
.......
[1644212595.921311][1076884:1076889] CHIP:DMG: ReportDataMessage =
[1644212595.9 [... cut]
closed 2022-02-07 by yufengwangca