project-chip/connectedhomeip #10433: there  exists a very subtle case where this will break if a subscriber over PASE tries to use keepSubscriptions=false.  Specifically, I think this would likely break when the PASE session becomes associated with a specific fabric after the addNOC command.

yunhanw-google, 2021-10-13:
FWIW, there is probably a very subtle case where this will break if a subscriber over PASE tries to use keepSubscriptions=false.  Specifically, I think this would likely break when the PASE session becomes associated with a specific fabric after the addNOC command.

If so (and if we desired to fix such a thing), we would need logic to consider the abstract concept of a peer node, which is indeed fabric index + node ID for CASE, but likely the session itself for PASE.

_Originally posted by @msandstedt in https://github.com/project-chip/connectedhomeip/pull/10339#discussion_r727586471_

mrjerryjohns, 2021-11-19:
We don't permit subscriptions over PASE, so punting this from TE7.5

closed 2021-11-19 by mrjerryjohns

msandstedt, 2021-11-19:
> We don't permit subscriptions over PASE.

Well that pretty well solves it.  Thanks!