project-chip/connectedhomeip #10493: [TE6] Unable to allocate. `UnauthenticatedSession exhausted`

doublemis1, 2021-10-14:
#### Problem
Issue found on TE6 branch.
The Accessory is not able to allocate data during subsequent commissions.
In device logs, the `UnauthenticatedSession exhausted` is present.
Because of that, the 4th controller commissioning fails.
As workaround;
1. controller should close the session 
2. reset accessory
3. controller resolve accessory, once again
4. controller can open commissioning window
5. next controller is able to commission accessory

praveenCY, 2021-10-18:
[](url)Observing this problem with TC-MF-1.13 Test Case

Step 1 : TH1_CR Commission DUT_CE : works
Step 2 : DUT_CE Commissioned to TH_CR3 : works
Step 3:  TH_CR2 try to commission DUT_CE : fails (open commission command is successful)

[log1.txt](https://github.com/project-chip/connectedhomeip/files/7367853/log1.txt)

Commissioning Issue with Third Controller is not seen with below changes, but any attribute read from DUT Fails after that. (log : [TC-MF-1_13_TICKET.txt](https://github.com/project-chip/connectedhomeip/files/7367879/TC-MF-1_13_TICKET.txt))

--- a/src/lib/core/CHIPConfig.h
+++ b/src/lib/core/CHIPConfig.h
@@ -2260,7 +2260,7 @@
 #ifndef CHIP_CONFIG_UNAUTHENTICATED_CONNECTION_POOL_SIZE
-#define CHIP_CONFIG_UNAUTHENTICATED_CONNECTION_POOL_SIZE 4
+#define CHIP_CONFIG_UNAUTHENTICATED_CONNECTION_POOL_SIZE 10

kghost, 2021-10-19:
#10652 is trying to increase the pool size. I'm not sure what triggers the second error.

Garik098, 2021-10-19:
On the commissioner side tests, TC-MF-1.14 and TC-MF-1.16 is hitting this same issue.

chulspro, 2021-10-22:
Can you please check this issue also? because we have got #13 step failed with TE6.

PR #11325 (2021-11-03): Fix UnauthenticatedSession leak (#10754)
#### Problem
Fix #10493 for TE6.LTS

#### Change overview
Backport #10754 to TE6.LTS

#### Testing
Verifed using unit-tests