project-chip/connectedhomeip #10689: [TE6][TC-IDM-4.1] SubscriptionRequestMessage from DUT test cases

younghak-hwang, 2021-10-20:
DUT : TV-App
Controller : Python Controller and ChipTool

 #### Problem
TV-App couldn't send report message.

 #### Proposed Solution
TV-App reports data
[Full_log_After_Reboot_and_Delete_tmp_file.txt](https://github.com/project-chip/connectedhomeip/files/7378927/Full_log_After_Reboot_and_Delete_tmp_file.txt)
[Full_Log_Chip_Tool.txt](https://github.com/project-chip/connectedhomeip/files/7378928/Full_Log_Chip_Tool.txt)
[Full_log_ChipTool_After_Reboot_and_Delete_tmp_file.txt](https://github.com/project-chip/connectedhomeip/files/7378929/Full_log_ChipTool_After_Reboot_and_Delete_tmp_file.txt)
[Full_Log_Python_Controller.txt](https://github.com/project-chip/connectedhomeip/files/7378930/Full_Log_Python_Controller.txt)
[Full_log_TV_App_After_Reboot_and_Delete_tmp_file.txt](https://github.com/project-chip/connectedhomeip/files/7378931/Full_log_TV_App_After_Reboot_and_Delete_tmp_file.txt)
[Full_log_TV_App_with_ChipTool_After_Reboot_and_Delete_tmp_file.txt](https://github.com/project-chip/connectedhomeip/files/7378932/Full_log_TV_App_with_ChipTool_After_Reboot_and_Delete_tmp_file.txt)
[Full_Log_TV-App_with_ChipTool.txt](https://github.com/project-chip/connectedhomeip/files/7378933/Full_Log_TV-App_with_ChipTool.txt)
[Full_Log_TV-App_with_PythonController.txt](https://github.com/project-chip/connectedhomeip/files/7378934/Full_Log_TV-App_with_PythonController.txt)
.

bzbarsky-apple, 2022-06-27:
@younghak-hwang Is this still an issue?  If so, can you please list clear steps to reproduce the problem?

younghak-hwang, 2022-06-28:
@bzbarsky-apple 
Today, I did a TEST based on the master branch.
Subscription TEST has been confirmed to work well.
We tested the chiptool cmd below.
The on-time attribute (0x00000401) currently defined in TC-IDM4.1 is not supported by TV-App, so we tested it with onoff attribute (0x0000).
I will close this.
--------------------------------------------------------------------------------------------------------------------------------
log
chip-tool onoff  subscribe on-off 0 100 1 1
...
[1656375052.544073][1278008:1278013] CHIP:TOO: Sending SubscribeAttribute to:
[1656375052.544147][1278008:1278013] CHIP:IN: Prepared secure message 0x56317d92b408 to 0x0000000000000001 (1)  of type 0x3 and protocolId (0, 1) on exchange 54038i with MessageCounter:189595420.
[1656375052.544167][1278008:1278013] CHIP:IN: Sending encrypted msg 0x56317d92b408 with MessageCounter:189595420 to 0x0000000000000001 (1) at monotonic time: 000000009026F878 msec
[1656375052.544222][1278008:1278013] CHIP:DMG: MoveToState ReadClient[0x7f3300004dc0]: Moving to [AwaitingIn]
[1656375052.554845][1278008:1278013] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:145895678 on exchange 54037i
[1656375052.554856][1278008:1278013] CHIP:EM: Found matching exchange: 54037i, Delegate: (nil)
[1656375052.554869][1278008:1278013] CHIP:EM: Rxd Ack; Removing MessageCounter:148592301 from Retrans Table on exchange 54037i
[1656375052.554875][1278008:1278013] CHIP:EM: Removed CHIP MessageCounter:148592301 from RetransTable on exchange 54037i
[1656375052.555280][1278008:1278013] CHIP:EM: Received message of type 0x5 with protocolId (0, 1) and MessageCounter:137317901 on exchange 54038i
[1656375052.555287][1278008:1278013] CHIP:EM: Found matching exchange: 54038i, Delegate: 0x7f3300004dc0
[1656375052.555293][1278008:1278013] CHIP:EM: Rxd Ack; Removing MessageCounter:189595420 from Retrans Table on exchange 54038i
[1656375052.555298][1278008:1278013] CHIP:EM: Removed CHIP MessageCounter:189595420 from RetransTable on exchange 54038i
[1656375052.555318][1278008:1278013] CHIP:DMG: ReportDataMessage =
[1656375052.555325][1278008:1278013] CHIP:DMG: {
[1656375052.555331][1278008:1278013] CHIP:DMG:  SubscriptionId = 0x997e448d,
[1656375052.555337][1278008:1278013] CHIP:DMG:  AttributeReportIBs =
[1656375052.555346][1278008:1278013] CHIP:DMG:  [
[1656375052.555352][1278008:1278013] CHIP:DMG:          AttributeReportIB =
[1656375052.555361][1278008:1278013] CHIP:DMG:          {
[1656375052.555367][1278008:1278013] CHIP:DMG:                  AttributeDataIB =
[1656375052.555373][1278008:1278013] CHIP:DMG:                  {
[1656375052.555380][1278008:1278013] CHIP:DMG:                          DataVersion = 0xd4c1a1f,
[1656375052.555387][1278008:1278013] CHIP:DMG:                          AttributePathIB =
[1656375052.555394][1278008:1278013] CHIP:DMG:                          {
[1656375052.555401][1278008:1278013] CHIP:DMG:                                  Endpoint = 0x1,
[1656375052.555407][1278008:1278013] CHIP:DMG:                                  Cluster = 0x6,
[1656375052.555415][1278008:1278013] CHIP:DMG:                                  Attribute = 0x0000_0000,
[1656375052.555421][1278008:1278013] CHIP:DMG:                          }
[1656375052.555429][1278008:1278013] CHIP:DMG:
[1656375052.555436][1278008:1278013] CHIP:DMG:                          Data = true,
[1656375052.555443][1278008:1278013] CHIP:DMG:                  },
[1656375052.555451][1278008:1278013] CHIP:DMG:
[1656375052.555456][1278008:1278013] CHIP:DMG:          },
[1656375052.555464][1278008:1278013] CHIP:DMG:
[1656375052.555469][1278008:1278013] CHIP:DMG:  ],
[1656375052.555478][1278008:1278013] CHIP:DMG:
[1656375052.555484][1278008:1278013] CHIP:DMG:  InteractionModelRevision = 1
[1656375052.555489][1278008:1278013] CHIP:DMG: }
[1656375052.555547][1278008:1278013] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0006 Attribute 0x0000_0000 DataVersion: 223091231
[1656375052.555557][1278008:1278013] CHIP:TOO:   OnOff: TRUE
[1656375052.555578][1278008:1278013] CHIP:DMG: MoveToState ReadClient[0x7f3300004dc0]: Moving to [AwaitingSu]
[1656375052.555592][1278008:1278013] CHIP:EM: Piggybacking Ack for MessageCounter:137317901 on exchange: 54038i
[1656375052.555606][1278008:1278013] CHIP:IN: Prepared secure message 0x56317d92b3e8 to 0x0000000000000001 (1)  of type 0x1 and protocolId (0, 1) on exchange 54038i with MessageCounter:189595421.
[1656375052.555616][1278008:1278013] CHIP:IN: Sending encrypted msg 0x56317d92b3e8 with MessageCounter:189595421 to 0x0000000000000001 (1) at monotonic time: 000000009026F884 msec
[1656375052.555830][1278008:1278013] CHIP:EM: Received message of type 0x4 with protocolId (0, 1) and MessageCounter:137317902 on exchange 54038i
[1656375052.555837][1278008:1278013] CHIP:EM: Found matching exchange: 54038i, Delegate: 0x7f3300004dc0
[1656375052.555845][1278008:1278013] CHIP:EM: Rxd Ack; Removing MessageCounter:189595421 from Retrans Table on exchange 54038i
[1656375052.555849][1278008:1278013] CHIP:EM: Removed CHIP MessageCounter:189595421 from RetransTable on exchange 54038i
[1656375052.555859][1278008:1278013] CHIP:DMG: SubscribeResponseMessage =
[1656375052.555866][1278008:1278013] CHIP:DMG: {
[1656375052.555870][1278008:1278013] CHIP:DMG:  SubscriptionId = 0x997e448d,
[1656375052.555876][1278008:1278013] CHIP:DMG:  MaxInterval = 0x64,
[1656375052.555881][1278008:1278013] CHIP:DMG:  InteractionModelRevision = 1
[1656375052.555889][1278008:1278013] CHIP:DMG: }
[1656375052.555896][1278008:1278013] CHIP:DMG: Subscription established with SubscriptionID = 0x997e448d MinInterval = 0s MaxInterval = 100s Peer = 01:0000000000000001
[1656375052.555905][1278008:1278013] CHIP:DMG: MoveToState ReadClient[0x7f3300004dc0]: Moving to [Subscripti]
[1656375052.555923][1278008:1278013] CHIP:DMG: Refresh LivenessCheckTime for 125000 milliseconds with SubscriptionId = 0x997e448d Peer = 01:0000000000000001

closed 2022-06-28 by younghak-hwang