project-chip/connectedhomeip #10759: EFR32 lighting-app Error resolving node id and two Leaders in one Thread network partition

Minan-Tuyasmart, 2021-10-21:
@jmartinez-silabs
**Problem：**
EFR32 lighting-app Error resolving node id and two Leaders in one Thread network partition

**Steps to reproduce the problem:**
platform:

- OTBR base on  matter_te6 branch (17cdb3b).

- Python Controller base on test_event_6 branch (82bcd4e).

- lighting-app run on the EFR32MG24 board base on commit 82bcd4e36f4856a4216cb76a16116faeec40e223

- Testing image on RasPi：[raspi_test_event_6_shrunk.zip](https://drive.google.com/file/d/1HWF8JAQ8HEPTi-Hx3Ds9CFiPTUCYvX6x/view)

​ 1. run command as follow on OTBR
`ubuntu@matter-te-6:~$ cd ot-br-posix/
ubuntu@matter-te-6:~/ot-br-posix$ sudo ot-ctl
>  factoryreset

> prefix add fd11:22::/64 paros
Done
> thread stop
Done
> ifconfig down
Done
> ifconfig up
Done
>  thread start
Done
> ipaddr
fdde:ad00:beef:0:0:ff:fe00:fc11
fdde:ad00:beef:0:0:ff:fe00:fc38
fdde:ad00:beef:0:0:ff:fe00:fc10
fd11:22:0:0:7449:ff46:2c8d:76d7
fdde:ad00:beef:0:0:ff:fe00:fc00
fdde:ad00:beef:0:0:ff:fe00:1800
fdde:ad00:beef:0:1338:1e65:fb42:bbab
fe80:0:0:0:14fd:bb91:1f5d:6ab3
Done
> dataset active -x
0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef000005109fc0721b1de8715fa942215d4592da48030a4f70656e54687265616401029754041021607808868aca08f58a5f57c99b0e0a0c0402a0fff8
Done
>  extpanid
dead00beef00cafe
Done`
2.run Python Controller
`(python_env) ubuntu@matter-te-6:~/connectedhomeip$ chip-device-ctrl
[1634804577.618333][178609:178609] CHIP:CTL: SyncSetKeyValue on ExampleOpCredsCAKey
[1634804577.618648][178609:178609] CHIP:CTL: SyncSetKeyValue on ExampleOpCredsICAKey
[1634804577.618914][178609:178609] CHIP:CTL: Generating NOC
[1634804577.619248][178609:178609] CHIP:CTL: Generating ICAC
[1634804577.619494][178609:178609] CHIP:CTL: Generating RCAC
[1634804577.619739][178609:178609] CHIP:CTL: SyncSetKeyValue on ExampleCARootCert0
[1634804577.625469][178609:178609] CHIP:IN: UDP::Init bind&listen port=0
[1634804577.625639][178609:178609] CHIP:IN: UDP::Init bound to port=44660
[1634804577.625674][178609:178609] CHIP:IN: UDP::Init bind&listen port=0
[1634804577.625783][178609:178609] CHIP:IN: UDP::Init bound to port=38376
[1634804577.625811][178609:178609] CHIP:IN: TransportMgr initialized
[1634804577.625835][178609:178609] CHIP:DIS: Init fabric pairing table with server storage
[1634804577.630183][178609:178609] CHIP:DL: Avahi client registered
[1634804577.631413][178609:178609] CHIP:ZCL: Using ZAP configuration...
[1634804577.631501][178609:178609] CHIP:ZCL: deactivate report event
[1634804577.632148][178609:178609] CHIP:DIS: Verifying the received credentials
[1634804577.634402][178609:178609] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
[1634804577.634452][178609:178609] CHIP:DIS: Assigned compressed fabric ID: 0x17FAF195FFAFCBEB, node ID: 0x000000000001B669
[1634804577.634487][178609:178609] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x17FAF195FFAFCBEB
[1634804577.652580][178609:178616] CHIP:DL: Platform main loop started.
Chip Device Controller Shell
`
[... 1555 lines cut ...]
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] HandleTXcharCCCDWrite - Config Flags value : 0
00> <info  > [DL] CHIPoBLE unsubscribe received
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEUnsubscribe
00> <detail> [IN] Clearing BLE pending packets.
00> <info  > [BLE] Releasing end point's BLE connection back to application.
00> <info  > [DL] Disconnect Event for handle : 1
00> <info  > [DL] BLE GATT connection closed (con 1, reason 4115)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEConnectionError
00> <detail> [DIS] OnExpiration - valid time out
00> <info  > [DL] OnSrpClientNotification: Last requested operation completed successfully
00> <info  > [DL] OnSrpClientNotification: Last requested operation completed successfully
00> <detail> [DIS] Using Thread extended MAC for hostname.
00> <info  > [DIS] Advertise operational node 1358DAE29DC5C9CC-0000000000BC5C01
00> <info  > [DL] advertising srp service: 1358DAE29DC5C9CC-0000000000BC5C01._matter._tcp
00> <detail> [DIS] OnDiscoveryExpiration callback for valid session
00> <error > [DIS] Failed to get extended timeout configuration err: Error CHIP:0x000000A0
00> <detail> [DIS] GetExtendedDiscoveryTimeoutSecs -1
00> <detail> [DIS] Using Thread extended MAC for hostname.
00> <info  > [DIS] Advertise commission parameter vendorID=9050 productID=21324 discriminator=3840/00
00> <info  > [DL] advertising srp service: 5B043129F6468CCF._matterc._udp
00> <error > [DIS] Failed to get extended timeout configuration err: Error CHIP:0x000000A0
00> <detail> [DIS] GetExtendedDiscoveryTimeoutSecs -1`

jmartinez-silabs, 2021-10-21:
Hi @Minan-Tuyasmart, I think it is just copy paste error in your manipulation

If we compare the thread operational dataset of you OTBR and the one provided in the command they are different

```
> dataset active -x
0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef000005109fc0721b1de8715fa942215d4592da48030a4f70656e54687265616401029754041021607808868aca08f58a5f57c99b0e0a0c0402a0fff8
Done 
```

`zcl NetworkCommissioning AddThreadNetwork 12344321 0 0 operationalDataset=hex:0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef00000510b2ef5d173d49db8c43d72476c9305c85030a4f70656e5468726561640102251f04106be48fb712106b08b7b7741c5b7521720c0402a0fff8 breadcrumb=0 timeoutMs=1000`

1 - 0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef00000510 
 **_9fc0721b1de8715fa942215d4592da48_**            030a4f70656e5468726561640102 
 **_9754041021607808868aca08f58a5f57c99b0e0a0_**            c0402a0fff8

2 - 0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef00000510 **_b2ef5d173d49db8c43d72476c9305c85_**                   030a4f70656e5468726561640102   **_251f04106be48fb712106b08b7b7741c5b7521720_**                c0402a0fff8

I highlighted here how there are different.

jmartinez-silabs, 2022-05-12:
I haven't heard anything back from this situation. It is an 8 months old ticket. I know Tuya has been able to complete recent test events. I am closing the bug