project-chip/connectedhomeip #10811: Some of the test cases have conditional PICS checks for each of the steps , need YAML support for this

raju-apple, 2021-10-21:
Look at this example in the TC-DM-1.1 test case : 
[image]

We need a way to conditionally execute certain steps based on the value , need a way in YAML to implement this as well

raju-apple, 2021-11-02:
https://github.com/project-chip/connectedhomeip/pull/11322

raju-apple, 2021-11-05:
PR Merged

kidip, 2021-11-05:
Initial PR is most likely not sufficient for the Test Harness.  Will work with @vivien-apple to design something that works for both CI and Test Harness.

vivien-apple, 2021-11-12:
Should have been fixed by #11729. Feel free to reopen if that still not enough.

closed 2021-11-12 by vivien-apple