project-chip/connectedhomeip #11262: chip-device-ctrl: attribute subscribe failed with "CHIP Error 0x0000000B: No memory"

Kxuan, 2021-11-01:
#### Problem
The test case `TC-IDM-4.2` of TE6 failed.

 * expected behavior
 `chip-device-ctrl` works as expected in the `TE6_TestPlanVerificationSteps-v1.3`.
 * actual behavior
 `chip-device-ctrl` failed at Step 16.
The error message shown in the TH console is:
```
chip-device-ctrl > zclconfigure OnOff OnOff 12344321 1 10 20 0
An exception occurred during configuring reporting of ZCL attribute:
CHIP Error 0x0000000B: No memory
```
Acturally, there is about 6.8GiB memory available on the rasperry pi. 

 * steps to reproduce
See `TC-IDM-4.2`
 * system configuration
  DUT:
     * ESP32
   TH:
     * Raspberry Pi 4 Model B Rev 1.4
     * RAM: 8GiB
     * SDcard: 32GiB

 * screenshots, images, logs, etc.
Logs: 
[TC-IDM-4.2.txt](https://github.com/project-chip/connectedhomeip/files/7452583/TC-IDM-4.2.txt)

andy31415, 2022-01-30:
No memory is likely related to pre-allocated number of concurrent subscriptions rather than actual RAM. The description says "subscribe 4-5 times" which leads me to believe that we never unsubscribe.

This seems to work as expected - we have a limit on number of concurrent subscriptions as each subscription take RAM on constrained devices (and those pre-allocate fixed sizes).

Asking for the linux-side to have higher limits seems reasonable (linux/mac/android can allocate a lot more RAM for subscriptions), changing device side to support more subscriptions is probably not possible and this is the reason we have wildcard subscriptions.

andy31415, 2022-01-30:
Tagging as "future" for now as this seems expected behaviour, however if we need to adjust subscribe prealloc count we can revisit.

closed 2022-07-30 by Kxuan