project-chip/connectedhomeip #11280: [YML/CHIP-Tool] Need a way to specify certain steps in YML should only be done in Test Harness or CI

kidip, 2021-11-01:
#### Problem
Right now there is no way in YML to say something should be run on the Test Harness, but not in CI (or vice versa).  We should have a way to do that.  We also should provide a build flag that lets us build in those steps or not depending on if we're building for CI or the Test Harness

raju-apple, 2021-11-02:
Ken , we discussed this yesterday , 2 options : 
1. If we don't even want to generate code , we could just leave the specific .yml file out of tests.js file and it will also not run in CI
2. If we want the code to be generated to be used by TH but don't want to run in CI , the only thing we need to do is to have the yaml file not start with Test_***, we could have something like Harness_**** so that those files would also not need to run in CI.

kidip, 2021-11-02:
I think option 2 is the better one.

@raju-apple do we have any tests where a specific step would be completely different?

raju-apple, 2021-11-02:
For now , we don't still have any scripts which are only for TH and not for CI , but currently for any steps we have in CI which needn't run, we are just using the `disabled` tag to skip those.

kidip, 2021-11-02:
Summarizing talk on Slack:  Looks like user prompts may need some revisiting and may be a separate issue.  This is still something we need, but might be lower priority for now.

closed 2022-10-01 by stale[bot]