project-chip/connectedhomeip #11281: Need a `FABRIC_IDX` type for ZAP XML

tcarmelveilleux, 2021-11-01:
#### Problem
Currently FabricIndex fields use `INT8U` everywhere, but the type should be FABRIC_IDX to make the semantic meaning clear (it's a fabric index), which would allow conditional processing in templates/helpers.

Example:
```
  <struct name="FabricDescriptor">
    <cluster code="0x003E"/>
    <item name="FabricIndex" type="INT8U"/>
    <item name="RootPublicKey" type="OCTET_STRING" length="65"/>
    <item name="VendorId" type="INT16U"/> <!-- Change INT16U to new type VendorID #2395 -->
    <item name="FabricId" type="FABRIC_ID"/>
    <item name="NodeId" type="NODE_ID"/>
    <item name="Label" type="CHAR_STRING" length="32"/>
  </struct>
```

bzbarsky-apple, 2021-11-01:
Using `FABRIC_IDX` or `fabric_idx` as the type should work fine right now....

bzbarsky-apple, 2021-11-01:
(I would vaguely prefer we use `fabric_idx`.)

tcarmelveilleux, 2021-11-01:
See: `src/app/zap-templates/zcl/data-model/chip/chip-types.xml`

```xml
<type id="0xF9" description="Fabric Index" name="fabric_idx" size="1" discrete="true" />
```

closed 2021-11-01 by tcarmelveilleux