project-chip/connectedhomeip #11286: Type `fabric_idx` is not used uniformly

tcarmelveilleux, 2021-11-01:
#### Problem
Many fabric-scoped structs use `INT8U` type instead of `fabric_idx`. 

 #### Proposed Solution
We need to audit and update all such uses to type `fabric_idx` which is already supported.

bzbarsky-apple, 2022-01-07:
Because we use the type to detect fabric-scoped things, this is a critical security issue.

tehampson, 2022-06-07:
I was asked to check if we have any INT8U in XML items that are for FabricIndex.

tl;dr other than some xmls in `src/app/zap-templates/zcl/data-model/` which I will look into further it seems like all FabricIndex in xml is using `fabric_idx`

-------------------------
I did: `find -name *.xml -not -path "./third_party/*" | xargs grep -i fabric.*index | grep -v type=\"fabric_idx\"` in chip repo and got :

```
./src/app/zap-templates/zcl/data-model/chip/test-cluster.xml:    <item name="fabricIndex" type="INT64U"/>
./src/app/zap-templates/zcl/data-model/chip/operational-credentials-cluster.xml:    <item name="InvalidFabricIndex" value="0x0b"/>
./src/app/zap-templates/zcl/data-model/chip/operational-credentials-cluster.xml:      <description>This command is used by Administrative Nodes to remove a given fabric index and delete all associated fabric-scoped data.</description>
./src/app/zap-templates/zcl/data-model/chip/binding-cluster.xml:    <item fieldId="0xFE" name="FabricIndex" type="FABRIC_IDX"/>
./src/app/zap-templates/zcl/data-model/chip/chip-types.xml:    <type id="0xF9" description="Fabric Index"             name="fabric_idx"        size="1"  discrete="true" />
```

bzbarsky-apple, 2022-06-08:
@tehampson The test-cluster bit needs to be fixed to use the right type _and_ the right field id (0xFE).

The others all look good.