project-chip/connectedhomeip #11293: iOS Framework integration dynamic vs static

amaolll123, 2021-11-01:
#### Problem
Currently, Amazon is working on integrating CHIP.framework into Alexa iOS Mobile App

However, as the built artifact of [CHIP framework](https://github.com/project-chip/connectedhomeip/tree/master/src/darwin/Framework) goes in `Dynamic Library` manner, it brings heavy latency impact on our app.

Question:
Is there a way to change the output format into `Static Library`? 
What would the workload look like for converting dynamic to static? 

@sagar-apple @vivien-apple

amaolll123, 2021-11-02:
I tried to change `Mach-o` type to `staticlib`, but the building will fail to 
```
"error: unsupported Mach-O type 'staticlib' for Text-Based API (in target 'CHIP' from project 'CHIP')"
```

amaolll123, 2021-11-08:
Turns out building in Xcode 13.1 will fix the issue. Closing.

closed 2021-11-08 by amaolll123

JourneyYoung, 2022-08-26:
Hello, I just change the output format into Static Library, and building in Xcode 14.0. But when I use Static Matter.framework in app target，It will be crash. Did you meet this issue?