project-chip/connectedhomeip #11295: Some of the test cases have dependency on Featuremap attribute, need chip-tool support

Garik098, 2021-11-01:
Few clusters have dependency on Featuremap attribute and the way we test the test case (Like the next steps in the test case) is changing with the value of the Featuremap attribute. Hence from the automation point of view, we need chip tool support for Featuremap attribute
[image]

woody-apple, 2022-01-25:
Can't this attribute be read?

raju-apple, 2022-01-26:
Looks like manually we have the ability to do this . @Garik098 can you double check please ?

Garik098, 2022-01-26:
Featuremap Attribute is now available in Chip tool. It was not present when I raised the ticket but it is now present to read. Closing the ticket.

closed 2022-01-26 by Garik098