project-chip/connectedhomeip #11296: Attributelist, commandlist, Eventlist, Featuremap attributes not present in Level Control cluster

Garik098, 2021-11-02:
As per the test case TC-LVL-1.1 we need to be able to read the Attributelist, commandlist, Eventlist, Featuremap attributes but they were currently not present in chip tool to read. Need these attributes to be added. 
[image]
. 
[image]

turon, 2022-02-01:
Level Control and Mode Selection clusters must comply with spec for v1.
Removing tag `v1_triage_split_4`

bzbarsky-apple, 2022-02-10:
Right now on tip:
```
 % ./out/debug/standalone/chip-tool levelcontrol read
[1644466330415] [32597:14894879] CHIP: [TOO] Missing attribute name
Usage:
  ./out/debug/standalone/chip-tool levelcontrol read attribute-name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Attributes:                                                                         |
  +-------------------------------------------------------------------------------------+
  | * current-level                                                                     |
  | * remaining-time                                                                    |
  | * min-level                                                                         |
  | * max-level                                                                         |
  | * current-frequency                                                                 |
  | * min-frequency                                                                     |
  | * max-frequency                                                                     |
  | * options                                                                           |
  | * on-off-transition-time                                                            |
  | * on-level                                                                          |
  | * on-transition-time                                                                |
  | * off-transition-time                                                               |
  | * default-move-rate                                                                 |
  | * start-up-current-level                                                            |
  | * server-generated-command-list                                                     |
  | * client-generated-command-list                                                     |
  | * attribute-list                                                                    |
  | * feature-map                                                                       |
  | * cluster-revision                                                                  |
  +-------------------------------------------------------------------------------------+
```
and
```
% ./out/debug/standalone/chip-tool modeselect read  
[1644466479878] [32643:14895975] CHIP: [TOO] Missing attribute name
Usage:
  ./out/debug/standalone/chip-tool modeselect read attribute-name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Attributes:                                                                         |
  +-------------------------------------------------------------------------------------+
  | * current-mode                                                                      |
  | * supported-modes                                                                   |
  | * on-mode                                                                           |
  | * start-up-mode                                                                     |
  | * description                                                                       |
  | * server-generated-command-list                                                     |
  | * client-generated-command-list                                                     |
  | * attribute-list                                                                    |
  | * cluster-revision                                                                  |
  +-------------------------------------------------------------------------------------+
```

which is just missing EventList.  That's tracked in https://github.com/project-chip/connectedhomeip/issues/15011 and might not be on track for 1.0, and in any case is not level control or mode select specific.

closed 2022-02-10 by bzbarsky-apple