project-chip/connectedhomeip #11309: Descriptor cluster returns all zeros in Device ID list

Damian-Nordic, 2021-11-02:
#### Problem
Descriptor cluster returns zeros for the DeviceList attribute:
```
​ chip-device-ctrl > zclread Descriptor DeviceList 1 0 0
--
...
[1634808840.922321][68589:68601] CHIP:ZCL: ReadAttributesResponse:
[1634808840.922343][68589:68601] CHIP:ZCL:   ClusterId: 0x0000_001D
[1634808840.922359][68589:68601] CHIP:ZCL:   attributeId: 0x0000_0000
[1634808840.922368][68589:68601] CHIP:ZCL:   status: Success                (0x0000)
[1634808840.922376][68589:68601] CHIP:ZCL:   attribute TLV Type: 0x16
[1634808840.922426][68589:68601] CHIP:ZCL:   attributeValue:
[1634808840.922438][68589:68601] CHIP:ZCL:   [
[1634808840.922448][68589:68601] CHIP:ZCL:     {
[1634808840.922454][68589:68601] CHIP:ZCL:       type: 0,
[1634808840.922461][68589:68601] CHIP:ZCL:       revision: 1,
[1634808840.922467][68589:68601] CHIP:ZCL:     },
[1634808840.922479][68589:68601] CHIP:ZCL:   ]

```

The value correspond to the following array defined in `endpoint_config.h`:
```
#define FIXED_DEVICE_IDS                                                                                                           \
    {                                                                                                                              \
        0, 0                                                                                                                       \
    }
```

 #### Proposed Solution
Find out what is missing in the ZAP code or the ZAP configuration.

tcarmelveilleux, 2022-01-31:
Repro steps obsolete, descriptor work done since then should have resolved. Closing as obsolete.

bzbarsky-apple, 2022-05-04:
Actually closing.

closed 2022-05-04 by bzbarsky-apple