project-chip/connectedhomeip #11335: CHIP Clusters do not support multiple requests in single message

krypton36, 2021-11-02:
#### Problem
Currently there is no way to implement tests or any tests covering the following tests in the steps where there is multiple requests in one message. The zzz_generated/controller-clusters/zap-generated/CHIPClusters.cpp file does not implement a way to perform these actions which the YAML test infrastructure uses to write tests. 

[[TC-IDM-3.2] Write Response Action from DUT to TH.](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-3-2-write-response-action-from-dut-to-th)
[[TC-IDM-4.1] SubscriptionRequestMessage from DUT test cases](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#351-tc-idm-41-subscriptionrequestmessage-from-dut-test-cases)
[[TC-IDM-4.2] ReportDataMessage from DUT test cases](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-4-2-reportdatamessage-from-dut-test-cases)

 #### Proposed Solution
Update the Zap files below to support the following:
Multiple AttributeRequests, EventRequests, and EventFilters in SubscribeRequestMessage and ReadRequestMessage. 
Multiple WriteRequests in WriteRequestMessage

src/app/zap-templates/templates/app/CHIPClusters.zapt
src/app/zap-templates/templates/app/CHIPClusters-src.zapt

 #### Notes
Once the proposed solution is added, the YAML framework should be able to add test cases based off the command added to the clusters. The parameters can be a list of EndPoints, and the value. Please take in consideration YAML framework. 

Example: [Test Link]() In step 2

raju-apple, 2022-01-26:
This is needed for 1.0

vivien-apple, 2022-02-02:
`CHIPClusters.zapt`  and `CHIPClusters-src.zapt` are gone but we will find some ways to get it to work.

raju-apple, 2022-02-23:
P1 - ETA 3/15

vivien-apple, 2022-03-03:
The command line bits (aka `chip-tool`) are here: https://github.com/project-chip/connectedhomeip/pull/15787
Those bits should let us remove the `manual-testing` tag from this issue.

closed 2022-09-22 by stale[bot]