project-chip/connectedhomeip #11338: Pump Configuration Control Cluster: operation-mode Write action not changing effective-operation-mode attribute value

liamgonyea, 2021-11-02:
#### Problem
When testing `[TC-PCC-2.2] Primary functionality with server as DUT`, running the `operation-mode` write command, the `effective-operation-mode` attribute should be updated but it does not change away from the default 0 value. 

Issue shows multiple times on Steps 2a, 3a, and 4a in `[TC-PCC-2.2] Primary functionality with server as DUT`.

#### Testing Setup
DUT: RasPi TE6 image's `chip-tool`
TH:  RasPi TE6 image's `chip-all-clusters-app`

[image]

Test Case: https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/pump_configuration_and_control.adoc#322-tc-pcc-22-primary-functionality-with-server-as-dut

DUT Logs attached.
[TC-PCC-2.2 Logs.txt](https://github.com/project-chip/connectedhomeip/files/7464170/TC-PCC-2.2.Logs.txt)

tlykkeberg-grundfos, 2022-04-21:
This should have been fixed with PR #17181

franck-apple, 2022-05-17:
Can be closed. Issue is resolved

closed 2022-05-17 by franck-apple