project-chip/connectedhomeip #11342: YAML support needed for multiple commissioners scenario

Garik098, 2021-11-02:
In Multi admin testcases, we need to run multiple commissioners (Chip tool, python controllers) at the same time connecting to DUT. Need YAML support to have multiple commissioners to automate these scenarios.
[image]

raju-apple, 2022-01-26:
P2 , manually possible to test . Only needed for scripting.

vivien-apple, 2022-01-27:
It exists already. You can use: `"identity": alpha` on any command and it will send the command from commissioner "alpha".

You can choose between:
  * alpha (default)
  * beta
  * gamma

closed 2022-01-27 by vivien-apple