project-chip/connectedhomeip #11343: Add YAML support for event verification when implemented in SDK

krypton36, 2021-11-02:
#### Problem
There are tests designed based off the specification which require verification of events. We need a way to verify events in the YAML framework to validate these tests. 

Tests that require Events: 
[[TC-DM-1.2] Basic Cluster Events](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicemanagement.adoc#312-tc-dm-12-basic-cluster-events)

[[TC-DIAGSW-1.2] Attributes with client as DUT](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/software_diagnostics.adoc#312-tc-diagsw-12-attributes-with-client-as-dut)

raju-apple, 2022-02-23:
P1 ETA 3/15

woody-apple, 2022-03-03:
Marking as P2, can be done manually presently.