project-chip/connectedhomeip #11352: Matter*PluginServerInitCallback are NOT guaranteed to be called only once

bzbarsky-apple, 2021-11-03:
#### Problem
The `Matter*PluginServerInitCallback` functions are implemented as if they are called once per process lifetime.  But that's not correct.  In particular, any time `DeviceController::Init` is called (which can happen during process lifetime due to shutting down an existing `DeviceController` and starting a new one), it will call `InitDataModelHandler`, which will call `emberAfInit (if USE_ZAP_CONFIG is defined along with use of DeviceController)`, which does `MATTER_PLUGINS_INIT`.  Which will call these callback functions try and register their attribute access callbacks.

Unfortunately, what that will lead to is loops in the linked list, and presumably infinite loops during attr reads after that.

 #### Proposed Solution
Sort out sane startup/shutdown semantics.  Presumably `InitDataModelHandler` should have a shutdown parallel that cleans up things as needed....

bzbarsky-apple, 2021-11-03:
@vivien-apple

bzbarsky-apple, 2021-11-03:
Or we should guard these "will only be called once" callbacks on something that actually ensures that in `emberAfInit`

mrjerryjohns, 2021-11-19:
Should this be targeted for TE7.5 @bzbarsky-apple ?

bzbarsky-apple, 2021-11-20:
I think this is a bugfix we can do after 7.5.