project-chip/connectedhomeip #11385: Support for feature maps

mrjerryjohns, 2021-11-03:
#### Problem

This is an umbrella issue tracking addition of basic support for feature maps to the SDK based on discussions with @bzbarsky-apple.

Namely, it entails:

1. Add the list of support features to the test cluster XML definition (likely as a `bitmap32` definition).
2. Modify the ZAP importer to ingest this information from the XML definition and automatically instantiate the 'feature map' attribute in the database.
3. Permit a user to select the right values for this attribute at configuration time for each cluster instance.
3. Emit a global attribute of type bitmap32 for each cluster instance in the generated descriptors/defaults/attribute store tables.

We should avoid any ZAP UI work as part of landing this basic version of the feature if we can.

mrjerryjohns, 2021-11-19:
As per discussions, we agreed that a basic, functional version of this already exists. So, punting this out of TE7.5 for now.

woody-apple, 2022-02-08:
@mrjerryjohns as I understand we need to add the feature map field to all relevant things, and this better SDK support can be post v1?

mrjerryjohns, 2022-08-15:
This is done.

closed 2022-08-15 by mrjerryjohns