project-chip/connectedhomeip #11393: [ota-provider-app] Make various values to QueryImageResponse Command configurable

ukatira, 2021-11-03:
#### Problem
The following fields of the QueryImageResponse command are currently not configurable.

The SoftwareVersion field of the QueryImageResponse command can be:
- Version of image being provided
- Incorrect value for testing purposes (Ex: TC-SU-2.2 Step-6)

UserConsentNeeded field of the QueryImageResponse command can be:
- False (default)
- True for testing purposes (Ex: TC-SU-2.2 Step-7)

 #### Proposed Solution
- Provide a way to respond with different SoftwareVersion field values
- Provide a way to respond with different UserConsentNeeded field values

dhrishi, 2022-01-28:
@shubhamdp PTAL

shubhamdp, 2022-02-01:
Command line option for user consent is added in PR #13816 and for software version it is added in #14619

PR #14619 (2022-02-07): ota-provider: cli options for software version, setup discriminator, and UserConsentNeeded fields
#### Problem
* Fixes #10923
* Fixes #11393

#### Change overview
* Support to set setup passcode on Darwin
* CLI options for software version and setup discriminator
* Way to respond with different UserConsentNeeded field values

#### Testing
* Manually tested on Darwin, various options of ota-provider-app/linux with chip-tool, ota-requestor-app/linux
closed 2022-02-07 by tcarmelveilleux