project-chip/connectedhomeip #11413: [ZAP] The generated defaults in `endpoint_config.h` does not comply with endianness

tlykkeberg-grundfos, 2021-11-04:
#### Problem
When setting a default value for a particular attribute in the .zap file, the resulting generated file `endpoint_config.h` does not reflect the big-endian and little-endian versions correctly. They are both stored big-endian wise. That would be expected to be different in the two `GENERATED_DEFAULTS` tables for big- and little-endian.

**Example:**
Setting the `LifetimeRunningHours` in the PCC cluster to `0x123456` in the ZAP UI tool, and then saving the .zap file, results in the following content in the .zap file. Which is correct.
```json
            {
              "name": "LifetimeRunningHours",
              "code": 21,
              "mfgCode": null,
              "side": "server",
              "included": 1,
              "storageOption": "RAM",
              "singleton": 0,
              "bounded": 0,
              "defaultValue": "0x123456",
              "reportable": 0,
              "minInterval": 1,
              "maxInterval": 65534,
              "reportableChange": 0
            },
```
Then running the `./scripts/tools/zap_regen_all.py` script generates the following contents in the `endpoint_config.h` file.

```c
#if BIGENDIAN_CPU
#define GENERATED_DEFAULTS                                                                                                         \
    {                                                                                                                              \
.......
            /* Endpoint: 1, Cluster: Pump Configuration and Control (server), big-endian */                                        \
                                                                                                                                   \
            /* 1377 - LifetimeRunningHours, */                                                                                     \
            0x12, 0x34, 0x56,                                                                                                      \
.......
#else // !BIGENDIAN_CPU
#define GENERATED_DEFAULTS                                                                                                         \
    {                                                                                                                              \
.......
            /* Endpoint: 1, Cluster: Pump Configuration and Control (server), little-endian */                                     \
                                                                                                                                   \
            /* 1377 - LifetimeRunningHours, */                                                                                     \
            0x12, 0x34, 0x56,                                                                                                      \
                                                                                                                                   \
.......
    }

#endif // BIGENDIAN_CPU
```

 #### Proposed Solution
Make the ZAP tool generate the correct default values in the `endpoint_config.h` files.

jmeg-sfy, 2021-11-04:
It is a know issues which block FeatureMap for example

jmeg-sfy, 2021-11-04:
https://github.com/project-chip/connectedhomeip/pull/9725

closed 2021-11-04 by bzbarsky-apple