project-chip/connectedhomeip #11454: Bridge-app has unmerged conflict on branch `test_event_7`

mikaelhm, 2021-11-04:
#### Problem
https://github.com/project-chip/connectedhomeip/blob/test_event_7/examples/bridge-app/linux/main.cpp#L450-L492
The `main.cpp` file for `bridge-app` has an unmerged conflict on branch `test_event_7`

 #### Proposed Solution
I don't know anything about the conflict, 

@andreilitvin, @pjzander-signify you seem to have been working on this, based on git blame.

cjandhyala, 2021-11-05:
Although we fix the merge conflicts and build the app, provisioning the app seem to fail. We need that to be looked at as well.

andy31415, 2021-11-05:
I believe I fixed it - at least I removed the merge markers and kept the two functions there.

andy31415, 2021-11-05:
I did not look into or resolve the 'provision the app fails'. For that please open a separate issue so we have only one per problem. That requires followup from someone familiar with the app (likely Paul).

closed 2021-11-05 by andy31415