project-chip/connectedhomeip #11491: provisioning of bridge app fails on raspi

cjandhyala, 2021-11-05:
with ToT (fb1a62eaa0755b8c28da6f5ed91087cfd81cf905), provisioning Bridage app on raspi fails with 

(process:18960): GLib-GIO-CRITICAL **: 18:07:09.704: g_dbus_proxy_call_sync_internal: assertion 'G_IS_DBUS_PROXY (proxy)' failed
[1636135629.705375][18960:18960] CHIP:DL: wpa_supplicant: failed to add network: Partyguys: unknown error
[1636135629.705447][18960:18960] CHIP:NP: Failed to connect to WiFi network: ../../examples/bridge-app/linux/third_party/connectedhomeip/src/platform/Linux/ConnectivityManagerImpl.cpp:1074: CHIP Error 0x000000AC: Internal error
[1636135629.705538][18960:18960] CHIP:DMG: ICR moving to [AddingComm]
[1636135629.705600][18960:18960] CHIP:DMG: ICR moving to [AddedComma]

Steps to reproduce.
1. Build raspi on raspi with  `scripts/examples/gn_build_example.sh examples/bridge-app/linux/ out/bridge-app`
2. Launch the app `sudo ./out/bridge-app/chip-bridge-app `
3. Build chip tool using `gn gen out/debug --args='chip_mdns="platform"'` and `ninja -C out/debug` , launch chip-tool with `sudo ./out/debug/chip-tool pairing ble-wifi <SSID> <PASSWD> 0 20202021 3840`

[chip_tool_log.txt](https://github.com/project-chip/connectedhomeip/files/7488105/chip_tool_log.txt)
[raspi_log.txt](https://github.com/project-chip/connectedhomeip/files/7488106/raspi_log.txt)

andy31415, 2021-11-08:
Could you confirm this is specific to the bridge app (i.e. do other example apps work correctly)?

cjandhyala, 2021-11-08:
@andy31415 Other apps provision fine with the same commit ID.

sihan6, 2021-11-16:
I have the same issue

lijinlei2011, 2021-11-19:
Hi，have the issue been resolved ?

ovidiu-usturoi, 2021-11-22:
Hello,  I have the same issue, but using  lighting-app on rpi (te7):

(process:3302): GLib-GIO-CRITICAL **: 21:58:27.028: g_dbus_proxy_call_sync_internal: assertion 'G_IS_DBUS_PROXY (proxy)' failed
[1637618307.028890][3302:3302] CHIP:DL: wpa_supplicant: failed to add network: matter_se: unknown error
[1637618307.028967][3302:3302] CHIP:NP: Failed to connect to WiFi network: ../../third_party/connectedhomeip/src/platform/Linux/ConnectivityManagerImpl.cpp:1074: CHIP Error 0x000000AC: Internal error

lijinlei2011, 2021-11-23:
Hi all,
the issue can be  resolved :

lijinlei2011, 2021-11-25:
Hi,
To be fair all of the reference apps have the same issue,you can resolve the issue by this way:

step 1: sudo ifconfig wlan0 down
step 2: killall wpa_supplicant
step 3: sudo ./chip- apps --wifi

utilize a Raspberry Pi 4 to execute applicable tests.

If the error is still existed， maybe you should change a place,because disturbance of the same frequency 

BRD

closed 2022-09-27 by stale[bot]