project-chip/connectedhomeip #11498: BSSID attribute read is not supported in wifi_network_diagnostics_server.cpp

praveenCY, 2021-11-05:
#### Problem
BSSID attribute read and related API link for platform are missing in wifi_network_diagnostics_server.cpp
Other attributes like channel_number, RSSI read is supported in the wifi_network_diagnostics_server.cpp properly

Missing GetWiFiBSSID(?) interface functions

praveenCY, 2021-11-17:
Looks like this support is merged as part of PR : https://github.com/project-chip/connectedhomeip/pull/11616 to master now
@yufengwangca : please close this if this is already supported

yufengwangca, 2022-02-07:
https://github.com/project-chip/connectedhomeip/pull/14755

yufengw@yufengw-SEi:~/connectedhomeip/out/debug/standalone$ ./chip-tool wifinetworkdiagnostics read bssid 12344321 0
.....
[1643913778.845776][878141:878146] CHIP:DMG: ReportDataMessage =
[1643913778.845800][878141:878146] CHIP:DMG: {
[1643913778.845804][878141:878146] CHIP:DMG: 	AttributeReportIBs =
[1643913778.845812][878141:878146] CHIP:DMG: 	[
[1643913778.845817][878141:878146] CHIP:DMG: 		AttributeReportIB =
[1643913778.845825][878141:878146] CHIP:DMG: 		{
[1643913778.845830][878141:878146] CHIP:DMG: 			AttributeDataIB =
[1643913778.845836][878141:878146] CHIP:DMG: 			{
[1643913778.845842][878141:878146] CHIP:DMG: 				DataVersion = 0x51ac8517,
[1643913778.845848][878141:878146] CHIP:DMG: 				AttributePathIB =
[1643913778.845856][878141:878146] CHIP:DMG: 				{
[1643913778.845862][878141:878146] CHIP:DMG: 					Endpoint = 0x0,
[1643913778.845869][878141:878146] CHIP:DMG: 					Cluster = 0x36,
[1643913778.845878][878141:878146] CHIP:DMG: 					Attribute = 0x0000_0000,
[1643913778.845884][878141:878146] CHIP:DMG: 				}
[1643913778.845892][878141:878146] CHIP:DMG: 					
[1643913778.845901][878141:878146] CHIP:DMG: 					Data = [
[1643913778.845910][878141:878146] CHIP:DMG: 						0x20, 0x2, 0x0, 0x0, 0x3, 0x56, 
[1643913778.845920][878141:878146] CHIP:DMG: 					]
[1643913778.845927][878141:878146] CHIP:DMG: 			},
[1643913778.845936][878141:878146] CHIP:DMG: 			
[1643913778.845942][878141:878146] CHIP:DMG: 		},
[1643913778.845950][878141:878146] CHIP:DMG: 		
[1643913778.845956][878141:878146] CHIP:DMG: 	],
[1643913778.845964][878141:878146] CHIP:DMG: 	
[1643913778.845970][878141:878146] CHIP:DMG: 	SuppressResponse = true, 
[1643913778.845976][878141:878146] CHIP:DMG: 	InteractionModelRevision = 1
[1643913778.845980][878141:878146] CHIP:DMG: }
[1643913778.846065][878141:878146] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0036 Attribute 0x0000_0000
[1643913778.846082][878141:878146] CHIP:TOO:   bssid: 200200000356

closed 2022-02-08 by woody-apple