project-chip/connectedhomeip #11506: PICS multiple conditionals in one step.

krypton36, 2021-11-05:
#### Problem
We need a way to define operators between multiple PICS values in one step. 

In Steps 3 and 4:
[Test Link](https://github.com/CHIP-Specifications/chip-test-plans/blob/5b213bdbf12bea3abea14aeb76bdc49df915e557/src/devicediscovery.adoc#test-procedure-10)

 #### Proposed Solution
Convert this code [here](https://github.com/project-chip/connectedhomeip/pull/11322/files?authenticity_token=pBmBGbbKgeaYTQNIeBDJUdnmQM%2BSqHja63zwYqZNa5OnwhyS556%2Bjr%2BaE6xc8RshutROjEXdteHx9%2FS6KydgWA%3D%3D&file-filters%5B%5D=.js#diff-2fc1f41e24c3d8a4616d4af7ff343574644b7ae86cec9bf18597258dfc79d94bR272) to the following:
```javascript
test[kPICSName].names.forEach((picsName, index) => {
    eval(picsName + '=' + PICS.get(picsName).value)
}
yaml.tests = yaml.tests.filter(test => test[kPICSName] == '' || eval(test[kPICSName].operation);
```
Example YAML:
```yaml
name: 3.2.2. [TC-DM-2.2] Operational Credential Attributes

config:
    cluster: "Operational Credentials"
    endpoint: 0

tests:
    - label: "Query NOCs"
      command: "readAttribute"
      attribute: "NOCs"
      response:
          constraints:
              type: list

    - label: "Query NOCs"
      command: "readAttribute"
      attribute: "NOCs"
      PICS: 
          names: [DUMMY_PICS0, DUMMY_PICS1, DUMMY_PICS2]
          operation: "!DUMMY_PICS0 || DUMMY_PICS1 && DUMMY_PICS2"
      response:
          value: [{NOC: "*"}]
          constraints:
              type: list
```

krypton36, 2021-11-05:
Please Note: Default Value will need to be updated as well.

vivien-apple, 2021-11-12:
Should have been fixed by #11729. Feel free to reopen if that still not enough.

closed 2021-11-12 by vivien-apple