project-chip/connectedhomeip #11508: OTA on TE7 branch has invalid Location param causing QueryImage on OTA-Provider to fail

liamgonyea, 2021-11-05:
#### Problem
When testing `TC-SU-3.1` I used 3 RasPis: OTA-Provider, chip-tool, OTA-Requestor.
Commissioning works fine, from chip-tool to both OTA-Provider and OTA-Requestor but attempting to run `./chip-tool otasoftwareupdaterequestor announce-ota-provider 1234 0 0 0 4321 0` was not successful as the `Location` parameter read by the OTA-Provider from QueryImage was not the correct length-2 "US" but instead "USZ???"

The same commands run on the master branch commit `cbf346e85d5e998d481ef137a4e5b240f2160ac3` does successfully run the commands but does not have the latest changes like the `test_event_7` branch.

#### Testing Setup
On `test_event_7` branch commit `c80fd0cfd732f764eafb5b6ffc66fd10e3e28ae6` for all 3 RasPis 

#### Failed Section of attached logs
`[1636151279.825767][1311532:1311532] CHIP:DMG: Received command for Endpoint=0 Cluster=0x0000_0029 Command=0x0000_0000
[1636151279.825821][1311532:1311532] CHIP:ZCL: OTA Provider received QueryImage
[1636151279.825851][1311532:1311532] CHIP:ZCL:   VendorID: 0x0
[1636151279.825878][1311532:1311532] CHIP:ZCL:   ProductID: 77
[1636151279.825906][1311532:1311532] CHIP:ZCL:   SoftwareVersion: 0
[1636151279.825933][1311532:1311532] CHIP:ZCL:   ProtocolsSupported: 0
[1636151279.825960][1311532:1311532] CHIP:ZCL:   HardwareVersion: 3
[1636151279.825988][1311532:1311532] CHIP:ZCL:   Location: USZ???
[1636151279.826016][1311532:1311532] CHIP:ZCL:   RequestorCanConsent: 0
[1636151279.826043][1311532:1311532] CHIP:ZCL:   MetadataForProvider: 0
[1636151279.826070][1311532:1311532] CHIP:ZCL: location param length 6 != expected length 2
[1636151279.826116][1311532:1311532] CHIP:DMG: ICR moving to [AddingComm]
[1636151279.826154][1311532:1311532] CHIP:DMG: ICR moving to [AddedComma]
[1636151279.826204][1311532:1311532] CHIP:EM: Piggybacking Ack for MessageCounter:0 on exchange: 27019r
[1636151279.826272][1311532:1311532] CHIP:IN: Prepared encrypted message 0xaaaab39d88a8 to 0x00000000000010E1 of type 0x9 and protocolId (0, 1) on exchange 27019r with MessageCounter:1.`

[TE7 TC-SU-3.1 Full Testing Logs (Fail).txt](https://github.com/project-chip/connectedhomeip/files/7489245/TE7.TC-SU-3.1.Full.Testing.Logs.Fail.txt)

liamgonyea, 2021-11-16:
### Update
Issue is still happening on latest latest changes using RasPi.

**However**, I am able to successfully run `./chip-tool otasoftwareupdaterequestor announce-ota-provider 1234 0 0 0 4321 0` after changing a line in the generated C++ code for the file `connectedhomeip/zzz_generated/ota-requestor-app/zap-generated/CHIPClusters.cpp`.

Roughly line 182 where it states:

>     SuccessOrExit(err = writer->PutString(TLV::ContextTag(argSeqNumber++), location.data()));

was changed to:
>     SuccessOrExit(err = writer->PutString(TLV::ContextTag(argSeqNumber++), location));

and the software img file now transfers successfully from OTA-P to OTA-R.

#### Test Setup
1 RasPi running OTA-Provider, chip-tool, and OTA-Requestor.
`test_event_7` commit: `f51eeb39de1e58c8f55a13cfc59782e8e11c6948`

#### Logs
I have put together both Failed Logs (similar to those from original post) and Pass Logs (after confirming several times that this was the only change made)

Fail Logs OTA-P snippet:

> [1637094679.965090][6380:6380] CHIP:DMG: Received command for Endpoint=0 Cluster=0x0000_0029 Command=0x0000_0000
[1637094679.965142][6380:6380] CHIP:ZCL: OTA Provider received QueryImage
[1637094679.965164][6380:6380] CHIP:ZCL:   VendorID: 0x0
[1637094679.965184][6380:6380] CHIP:ZCL:   ProductID: 77
[1637094679.965204][6380:6380] CHIP:ZCL:   SoftwareVersion: 0
[1637094679.965224][6380:6380] CHIP:ZCL:   ProtocolsSupported: 0
[1637094679.965243][6380:6380] CHIP:ZCL:   HardwareVersion: 3
[1637094679.965323][6380:6380] CHIP:ZCL:   Location: USÿœˇˇ
[1637094679.965347][6380:6380] CHIP:ZCL:   RequestorCanConsent: 0
[1637094679.965367][6380:6380] CHIP:ZCL:   MetadataForProvider: 0
[1637094679.965387][6380:6380] CHIP:ZCL: location param length 6 != expected length 2

[Fail Logs.zip](https://github.com/project-chip/connectedhomeip/files/7549854/Fail.Logs.zip)

Pass Logs OTA-P snippet:

> [1637095045.776068][6633:6633] CHIP:BDX: OutputEvent type: AckEOFReceived
[1637095045.776154][6633:6633] CHIP:BDX: Transfer completed, got AckEOF

[Pass Logs.zip](https://github.com/project-chip/connectedhomeip/files/7549855/Pass.Logs.zip)

carol-apple, 2021-11-17:
https://github.com/project-chip/connectedhomeip/pull/11661 is the fix that exists on master branch which addresses this issue.

carol-apple, 2022-02-02:
@liamgonyea I don't see this issue on ToT of master branch. I'll close this. Please file a new issue if you see it again.

closed 2022-02-02 by carol-apple