project-chip/connectedhomeip #11524: TE6 [TC-TSTAT-2.1] Some Thermostat Attributes are missing from YAML framework

manjunathambiger23, 2021-11-08:
Summary :
As per Spec there are 49 attributes which supports Thermostat Cluster but when made a run 22 attributes are missing in YAML framework.

Spec Screenshot and Log file :
[image]
[TSTAT_2_1.txt](https://github
[image]
.com/project-chip/connectedhomeip/files/7497106/TSTAT_2_1.txt)

chrisbeach59, 2021-11-22:
Only the attributes currently supported by the thermostat cluster are in the controller zap file.  The ones related to scheduling are present as scheduling implementation had been started, but were unable to be completed and tested due to missing functionality in the SDK (lists and structs).

closed 2021-11-22 by chrisbeach59

kowsisoundhar12, 2022-02-24:
@chrisbeach59 still observing attribute missing error in TC-TSTAT-2.1 Scripts,please find spec screenshot and error logs
Error screenshot :
[image]

Spec screenshot : 
[image]
[image]
[image]
[image]
[image]