project-chip/connectedhomeip #11548: SRP client to check if service record exists before adding service record

AlanLCollins, 2021-11-08:
#### Problem
Chip over openthread:
OnSrpClientNotification: Domain name or RRset duplicated 

How to replicate: 

1. setup CHIP Controller and CHIP light that use openthread stack. 
2. provision CHIP light to join thread network created by CHIP controller. 
3. Pair devices normally to be able to send Cluster commands, and read attributes. 
4. factory reset the CHIP-light (if using nRF-DK, you can long-press button 1. 
5. repeat steps 2 and 3, You will see the SRP server will get notification --> OnSrpClientNotification: Domain name or RRset duplicated 

openthread project recently added implementation to return a failure when trying to add new service that already exists: https://github.com/openthread/openthread/pull/6597 

 #### Proposed Solution
The SRP client should check if the service records exists, before trying to add new service.

abtink, 2021-11-09:
@AlanLCollins  

- OpenThread SRP client implementation reports back any error code it gets from the SRP server, including duplicate name error (in a callback).
  - The SRP client is not in charge of choosing the host and/or service instance name and it is up to next layer (module that is using its API, e.g. CHIP layer code) to pick the names and then handle any name conflicts/errors.
- I think your suggestion is that before a device tries to register a name, it checks to see if the name is registered by any other device. 
  - It is technically possible to perform a DNS-SD browse for a name before trying to register it.
  - But I would say it is not something SRP client should do and would be something next layer can decide to perform.
  - Also I think this would be somewhat inefficient, as the same info can be determined by trying to register the service using SRP client APIs and then getting the error (through the SRP client callback) that server rejected the registration and checking the error code for duplicated error (i.e., name conflict error).

-------

Some questions/comments on the behavior you mention:
- Is the same host name and/or service instance name(s) being registered by the CHIP-light after factory reset?
- If so, then the behavior of server/client seems to be correct.  
   - SRP server associates a registered host name and its service instance name with a device using the ECDSA public key that is used during registration of the service by a device.
  - After a successful registration, the SRP server would not allow any other device (with a different ECDSA SRP key) to claim the same host name and/or service instance name(s). This is enforced by server during the granted key-lease interval after a successful service registration.
   - When the CHIP device is factory reset it would behave as a new device and would create and use a new ECDSA SRP key pair for SRP registration.
   - So if the CHIP light then tries to register same host name and/or same instance name now with a new SRP key, the server should correctly reject it as duplicate and not allow it to register same names.
   
- The common solution/model is that the hostname and/or service instance names are derived from some node id which is changed upon factory reset (probably randomly chosen upon factory reset).

AlanLCollins, 2021-11-20:
Hello @abtink , my apologies for the delay. Today I was able to follow up on this topic. I have new information. 
* I confirmed the accessory device (nRF52840DK LIGHT) is successfully creating new srp service name after reset. Regardless if its a factory-reset or power-on reset:
- I: 785 [DL]advertising srp service: 5C3390334C7C9C1A._matterc._udp
- I: 789 [DL]advertising srp service: 16347930BA546E65._matterc._udp
- I: 783 [DL]advertising srp service: 786F5743B108C814._matterc._udp

* I confirmed the instance name is advertised correctly by the border router.
$ dns-sd -B _matterc._udp
Browsing for _matterc._udp
DATE: ---Fri 19 Nov 2021---
14:32:03.362  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
14:32:03.363  Add        3  10 local.               _matterc._udp.       16347930BA546E65
14:32:03.363  Add        3  10 local.               _matterc._udp.       786F5743B108C814
14:32:03.363  Add        2  10 local.               _matterc._udp.       5C3390334C7C9C1A

However, the CHIP SDK assigns new service to advertise operational node. That gets build from the **Fabric_id + node_id**:
I: 10563 [DIS]Advertise operational node DAF991072265D5FE-00000000009BAFA0
I: 10571 [DL]advertising srp service: DAF991072265D5FE-00000000009BAFA0._matter._tcp
E: 10578 [SVR]Operational advertising enabled

Therefore, if we trigger a new matter pairing/provisioning (which triggers a new service discovery) for a node that has changed the instance name but gets the previous node_id and fabric_id, it will generate a domain or RRset duplicate error. 

You should be able to reproduce by using nRF52840 LIGHT demo, and other CHIP-Controller or CHIP-Tool example. 
1) Start pairing using fabric_id= X, and node_id= Z, this should create a secure BLE session to later use methods from networkCommissioning cluster AddThreadNetwork and EnableThreadNEtwork. To invite the CHIP LIGHT to join your Thread network. 
2) Then start DeviceController::UpdateDevice to trigger the Mdns::resolver. 
3) The first attempt should results successful. Then Factory-reset the nRF52840DK board by long pressing button 1 (~10s)
4) repeat steps 1 and 2 using the same fabric_id and node_id.

wgtdkp, 2021-11-22:
@AlanLCollins 
>4. repeat steps 1 and 2 using the same fabric_id and node_id.

This doesn't seem a reasonable case for me. For what I know, `node_id` is designed to uniquely identify a Matter device on a fabric link. If a device is factory reset, it should be provisioned again with a different `node_id` because the device is now totally a new one to the CHIP commissioner. @tcarmelveilleux may confirm.

msandstedt, 2022-01-06:
> Therefore, if we trigger a new matter pairing/provisioning (which triggers a new service discovery) for a node that has changed the instance name but gets the previous node_id and fabric_id, it will generate a domain or RRset duplicate error.

I agree with @wgtdkp.

This is a bug in fabric administration and commissioning, not SRP.  A factory-reset node is effectively a new node, and a fabric administrator should not assign in-use node IDs to new a commissionee because this breaks the security model.

AlanLCollins, 2022-01-07:
Thank you for  your responses. I will close this ticket.

closed 2022-01-07 by AlanLCollins