project-chip/connectedhomeip #11552: BDX messages monitoring/processing should be done by Test Harness

ukatira, 2021-11-08:
#### Problem
Currently we are logging BDX messages on OTA Reference Apps to enable testing for TE7

[1636153795565] [83609:59400960] CHIP: [BDX] Sending BDX Message
[1636153795565] [83609:59400960] CHIP: [BDX] ReceiveInit
[1636153795565] [83609:59400960] CHIP: [BDX]   Proposed Transfer Control: 0x20
[1636153795565] [83609:59400960] CHIP: [BDX]   Range Control: 0x0
[1636153795565] [83609:59400960] CHIP: [BDX]   Proposed Max Block Size: 1024
[1636153795565] [83609:59400960] CHIP: [BDX]   Start Offset: 0x0000000000000000
[1636153795565] [83609:59400960] CHIP: [BDX]   Proposed Max Length: 0x0000000000000000
[1636153795565] [83609:59400960] CHIP: [BDX]   File Designator Length: 8
[1636153795565] [83609:59400960] CHIP: [BDX]   File Designator: test.txt

 #### Proposed Solution
Test Harness should inspect BDX packets that come through and perform verification at that level rather than parsing logs on OTA reference apps

holbrookt, 2022-02-01:
@ukatira Is this an issue for 1.0?

ukatira, 2022-02-01:
I'm okay if we continue having these log prints in OTA Reference Apps to enable verification for 1.0

closed 2022-10-12 by stale[bot]