project-chip/connectedhomeip #11554: Cache attribute values for peer Sleepy Devices

pan-apple, 2021-11-08:
#### Problem
This applies mainly for controller devices. If a peer device is a sleepy device, it will not be reachable at all times. If the controller application tries to read or write attributes of such devices, it'll currently fail.

To mitigate, the controller API can cache the attribute values for such devices. The read requests will return the last known value from the controller cache. The write attribute request will update the cache, and enqueue a command to be sent to the sleepy device.

mrjerryjohns, 2021-11-09:
I'm actually working on something similar caching attribute data on reads/subscribes such that there's a cache available that describes the latest state on the target device. We should likely collaborate.

mrjerryjohns, 2022-01-31:
A class to implement this (`AttributeCache`) has been implemented and checked in a while back

closed 2022-01-31 by mrjerryjohns