project-chip/connectedhomeip #11555: Implement mechanism to send commands reliably to sleepy devices

pan-apple, 2021-11-08:
#### Problem
Sleepy devices are not always available to receive and act on commands. We need to add a mechanism to matter SDK using which a non-sleepy device can  reliably send a command to a sleepy device. The following ideas were explored.

1. Queue up the commands on non-sleepy device for a peer sleepy device
2. Subscribe to reports from a sleepy device. When the reports are received, before acknowledging the report status, send the queued up commands to the device.

Alternate proposal is to keep retrying the commands to a sleepy device until it is successful.

This issue is to track the work needed for this functionality.

SkipAshton, 2021-11-29:
Can we close this item as the individual work items are covered by other items included in the list for sleepy devices?

turon, 2021-12-01:
Per discussion:

Per peer node queue is a general mechanism, but required for clients of sleepy nodes.
Don't close until implemented.

mrjerryjohns, 2022-01-11:
What does queuing mean in this context above and beyond what the stack already does today?

turon, 2022-01-12:
Agree that the exchange sequencing in the stack should naturally handle most cases.  We should validate though with some test cases:

1) Single client writing to sleepy with multi-second idle latency.
a)
- commission 2 node topology
- confirm sleepy is operational and goes in idle state
- send on/off to sleepy w/ 5 sec idle
- confirm client persists and waits the correct time, and transitions to active mode timing for some follow-up messaging on response

b) add variant to above where sleepy is offline before receiving first onoff command forcing client to perform retries @ idle timing

2) Multiple clients writing to sleepy with multi-second idle latency.

- Variation of test 1 but with two competing clients
- Confirm both clients retry as appropriate so they eventually succeed despite flushing of the limited SED buffers at the sleepy child's parent router.

pan-apple, 2022-01-13:
I would like to propose another test case. It's a single client sending multiple commands to a given sleepy device.

- commission 3 node topology: client node, parent thread router, and server sleepy node
- confirm sleepy is operational and goes into idle state
- send on/off to sleepy w/ 30 sec idle
- send read attribute command to the sleepy device on same or some other cluster, while the previous command is still pending
- confirm client persists, waits the correct time, and transitions to active mode timing for some follow-up messaging on response
- confirm the read attribute command returns the correct value

turon, 2022-01-18:
@pan-apple yes, that test case sounds good.  It basically enhances and can replace the case (1) I proposed.

woody-apple, 2022-02-28:
Assigning based on the dup.

mrjerryjohns, 2022-03-01:
I'm going to close out this issue as "nothing needs to be done" for v1.0, and shift the rest to actual test cases. Does that sound good @turon?

turon, 2022-03-01:
As discussed in sw dev call today, this issue was meant to track a centralized (powered) controller that subscribes to a sleepy sensor and is able to queue commands + attribute writes that get delivered once the sensor wakes and sends events.

turon, 2022-03-03:
Some relation to "optimistic write cache" of remote device state via attributes.

Extra considerations for commands based on whether they are omnipotent or not.  Caching commands that are not omnipotent can create issues.