project-chip/connectedhomeip #11568: Make ConnectivityMgr().SetDelegate(&gDiagnosticDelegate) survive through  CHIP stack is shut down and restarted

yufengwangca, 2021-11-09:
#### Problem
 
void MatterGeneralDiagnosticsPluginServerInitCallback()
{
    registerAttributeAccessOverride(&gAttrAccess);

    PlatformMgr().SetDelegate(&gDiagnosticDelegate);
    ConnectivityMgr().SetDelegate(&gDiagnosticDelegate);
 }

SetDelegate depends on being called once per process execution. It will not work right if it's called more than once as things stand. If the CHIP stack is shut down and restarted (e.g. via PlatformManager::Shutdown and then a new InitChipStack, etc), either this function will get called again, in which case the attribute access interface bit will be broken, or it will not be called again, in which case are these delegates going to be set up right?

 
Member
Author
@yufengwangca yufengwangca 2 hours ago
MatterGeneralDiagnosticsPluginServerInitCallback should be called only once during the lifetime of per process execution. Do we have the scenario we shut down the CHIP stack and restart the CHIP stack without restarting the process? If there is the case, then "registerAttributeAccessOverride(&gAttrAccess);" will also not get called, we are not registering the handle for these platform attributes.

 
Member
@bzbarsky-apple bzbarsky-apple 1 hour ago
MatterGeneralDiagnosticsPluginServerInitCallback should be called only once during the lifetime of per process execution.

That is the intent, I think... It happens to not be true in practice right now; see #11352

Do we have the scenario we shut down the CHIP stack and restart the CHIP stack without restarting the process?

Yes, absolutely.

If there is the case, then "registerAttributeAccessOverride(&gAttrAccess);" will also not get called,

Yes, but shutting down the stack does not unregister attr access overrides, so this part is fine.

But in general, we need to decide what the call frequency guarantees are around these callbacks and then make sure that everything in them is operating correctly given those frequency guarantees. Right now, if we actually change MatterGeneralDiagnosticsPluginServerInitCallback to be once-per-process-lifetime this code would break.
 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2021-11-09:
@vivien-apple

yufengwangca, 2022-02-28:
Currently, shutting down the stack does not unregister delegate, so after PlatformManager::Shutdown and then a new InitChipStack, the delegate is still there until another SetDelegate is called

yufengwangca, 2022-05-31:
https://github.com/project-chip/connectedhomeip/pull/16085

bzbarsky-apple, 2022-06-01:
I don't understand.  There were no code changes made in #16085.  @yufengwangca what am I missing?