project-chip/connectedhomeip #11572: Support reading element be index from attribute with list type

erjiaqing, 2021-11-09:
#### Problem
Currently, the read requests are implemented by AttributeAccessOverride, however, the implementation does not support encode a single element in the list and getting the length of the list.

This function is required by list chunking feature, which requires accessing a single element from the list.

 #### Proposed Solution
Update API to allow the AttributeAccessOverride return the length of list (if it is a list, maybe with a type info), and writing a single element by index.

bzbarsky-apple, 2021-11-10:
We can check for the attribute being a list via the attribute metadata in the attribute database.

So one option here is to do that lookup before we do the AttributeAccessOverride and then use that to decide how to do the list encoding.

At that point I feel like we have two options: we can either just always encode in one-element-per-chunk form (and rewrite existing implementations of Read to do that), or we could call into the normal Read() API, and if it comes back with `CHIP_ERROR_BUFFER_TOO_SMALL` we rewind the writer and fall back to item by item....

erjiaqing, 2021-11-10:
> We can check for the attribute being a list via the attribute metadata in the attribute database.

True

> So one option here is to do that lookup before we do the AttributeAccessOverride and then use that to decide how to do the list encoding.

To achieve this, we need to implement:
1. Get the length (in # of elements) of the attribute from AttributeAccessOverride
2. Encode a single element from the list by AttributeAccessOverride

> At that point I feel like we have two options: we can either just always encode in one-element-per-chunk form (and rewrite existing implementations of Read to do that), or we could call into the [...]

Of course, the second one looks better, however, 1 and 2 are more like two stage for implementing this to me.
I would suggest implement the encoding to break the list in to many elements first, and take 2 as a optimization later.

bzbarsky-apple, 2021-11-10:
> Get the length (in # of elements) of the attribute from AttributeAccessOverride

Or have some way for the "encode" call to indicate "index is at length of list", right?

> I would suggest implement the encoding to break the list in to many elements first, and take 2 as a optimization later.

In a vacuum, sure, but at the moment this would require rewriting all of our list getters, including ones that I suspect alway fit in an MTU, right?

erjiaqing, 2021-11-10:
> > Get the length (in # of elements) of the attribute from AttributeAccessOverride
> 
> Or have some way for the "encode" call to indicate "index is at length of list", right?

Acceptable

> > I would suggest implement the encoding to break the list in to many elements first, and take 2 as a optimization later.
> 
> In a vacuum, sure, but at the moment this would require rewriting all of our list getters, including ones that I suspect alway fit in an MTU, right?

Yes & No

Infact, most attribute should fit in a MTU for now, and ths issue is when we meet a list, we may already used some buffer so it cannot fit into the remaining buffer.

Is it possible to add another flag to the override to indicate if it supports list chunking, or add a internal list (so we can remove it later easily) for the attributes that don't list chunking (then new attributes should always support list chunking).

mrjerryjohns, 2021-11-10:
How about this suggestion - we still permit implementors to use the existing `AttributeValueEncoder::EncodeList` call to encode out the entire list, but internally within the encoder, we automatically manage chunking of list data, which includes writing out individual `AttributeDataIB` containing the right paths and so on. Those bits of context can be stashed away inside that object without consumers having to know.

More importantly, later, when we need to resume from a particular list index, or only retrieve a particular list item, that will still entail the same call to read out the list, but within the encoder, it will only _actually_ encode the right list items that is requested of it.

This actually might be simpler for consumers, since in addition to the work to changing out all the APIs, it is actually potentially more work to ask clients to retrieve a particular item as opposed to all items.

tcarmelveilleux, 2021-11-10:
I agree with @mrjerryjohns  on https://github.com/project-chip/connectedhomeip/issues/11572#issuecomment-965696646

I would suggest that on top of a "default" EncodeList that does the filtering by itself, we have an "explicit" version that is able to receive the flags for the read span, so that for usecases where the reads are very costly or there are large amount of data, and the server cluster logic wants to only read what is needed, rather than having post-filtering, it can be done:

```
aEncoder.EncodeListExplicit([](const TagBoundEncoder & encoder, FabricIndex filteredFabricIndex, IterationRange range) {
   // Do the reads here based on `range` and possibly filter by fabric
   // if filteredFabricIndex != 0.
}
```

bzbarsky-apple, 2021-11-10:
> How about this suggestion 

That sounds pretty plausible...  Basically change the actual object handed to the EncodeList lambda from TagBoundEncoder to something that knows to drop some number of writes and then start accepting them or something along those lines?  Plus the stopping/starting of AttributeDataIB.

> We'd need to create an intercept API to encode out a list element within AttributeValueEncoder itself

I'm not sure I'm following this part.

I agree that it might be good to have something where the server can opt in to "tell me which parts you _really_ want" as @tcarmelveilleux suggests.

mrjerryjohns, 2021-11-11:
> I'm not sure I'm following this part.

When you call `AttributeValueEncoder::EncodeList` today, it calls into `TagBoundEncoder::EncodeList` to actually do the starting of the list, then, calling the lambda with a reference to the `TagBoundEncoder` passed in. However, this is likely not the object we should pass in to handle this list chunking logic, since `TagBoundEncoder` has a fairly narrow, well-defined purpose.

What should be passed into the lambda is a reference to the `AttributeValueEncoder` itself, which needs to house a `EncodeListItem` method on it that will in turn under the covers, manage the chunking logic, only calling `DataModel::Encode` when it needs to.

Something like this in AttributeValueEncoder:

```
    template <typename... Ts>
    CHIP_ERROR EncodeListItem(Ts... aArgs) const
    {
        if (noMoreSpace) {
            return CHIP_NO_ERROR;
        }

        if (mListIndex == 0) {
            // Encode empty AttributeData IB
        }
        else {
            // Start an Attribute Data IB
            Encode(*mWriter, mTag, std::forward<Ts>(aArgs)...);
            // Close out Attribute Data IB
        }
    }

    template <typename ListGenerator>
    CHIP_ERROR EncodeList(ListGenerator aCallback)
    {
        mTriedEncode = true;
        if (mWriter == nullptr)
        {
            return CHIP_NO_ERROR;
        }

        aCallback(this);
    }
```

bzbarsky-apple, 2021-11-11:
> What should be passed into the lambda is a reference to the AttributeValueEncoder itself, which needs to house a EncodeListItem method on it that will in turn under the covers

Yes, ok, this makes total sense and matches what I was thinking, with one correction: it should NOT be the AttributeValueEncoder itself.  It should be a helper class that exposes only the APIs that are needed for doing the list items.  For example, we would not want to allow calling EncodeList on that object passed to the callback function....

This helper object can obviously hold a ref to the AttributeValueEncoder and so forth.

mrjerryjohns, 2022-02-01:
For the purposes of chunking, this has been implemented and tested.

closed 2022-02-01 by mrjerryjohns

turon, 2022-02-01:
Removing `v1_triage_split_4`.