project-chip/connectedhomeip #11625: TE6 [TC-TSTAT-2.1] Attribute default value mismatch

manjunathambiger23, 2021-11-10:
Summary:
As per Spec the default value of NumberOfWeeklyTransitions and NumberOfDailyTransitions Attribute needs to be 0 but when we mexecute YAML script the value returned for NumberOfWeeklyTransitions and NumberOfDailyTransitions are 7 and 4.

Spec screenshot :
[image]

Execution Screenshot:
[image]
[image]

Execution log :
[TSTAT_2_1.txt](https://github.com/project-chip/connectedhomeip/files/7511410/TSTAT_2_1.txt)

chrisbeach59, 2021-11-22:
The defaults are there to support dependencies for when an attribute is not present.  If you look at the zap file for the thermostat example.  That is the current values that are configured.  A DUT at power up is not expected/required to respond with default values.

closed 2021-11-22 by chrisbeach59

manjunathambiger23, 2022-02-03:
@chrisbeach59 as per the test plan they are asking us to check default value
[image]