project-chip/connectedhomeip #11632: Evaluate use of DNS cache

cecille, 2021-11-10:
#### Problem
It's not clear what the purpose of the cache is or how it should be sized - the TTL on most entries would be only 2 minutes and nodes should be querying only very specific nodes they want to connect to. Many larger devices will be using an mdns impl that already includes caching. See discussion on https://github.com/project-chip/connectedhomeip/pull/11149.

 #### Proposed Solution
I think there is utility in leaving the code there in case there are cases we have not considered, but we may want to consider turning it off by default.

andy31415, 2022-02-24:
Assigned to me as I believe I am touching this as part of mDNS resolution logic (I need to have a 'get best address' logic and the DNS caching is actually in the way in parts). Not working to really fix it, but it may be touched as a sideffect.

andy31415, 2022-06-03:
We removed the cache.

closed 2022-06-03 by andy31415