project-chip/connectedhomeip #11641: Expose Subscription liveness callbacks to the application layer

sagar-apple, 2021-11-10:
#### Problem
There doesn't seem to be a way for an application to sign up for `SubscriptionTerminated` notifications from the SDK. 

This means that if a subscription expires, no notification is sent to the application to attempt a retry, for example. 

 #### Proposed Solution
Expose the IMDelegate APIs to the application layer.

bzbarsky-apple, 2021-11-10:
@mrjerryjohns @yunhanw-google

mrjerryjohns, 2021-11-10:
I presume you're referring to the client side here.

There are a couple of different callback APIs in question here - starting at the base with `ReadClient::Callback`, terminal errors are delivered through `ReadClient::Callback::OnError`, while non-terminal, attribute-path specific errors are delivered through `ReadClient::Callabck::OnAttributeData`.

The next level up is `chip::Controller::TypedReadCallback` (which is passed into `chip::Controller::ReadAttribute`). The `OnErrorCallbackType` signature will convey terminal errors if `aPath == nullptr`, including those that are fatal to the subscription, with the actual error embedded in a combination of the IM status code + CHIP_ERROR provided.

What we don't do well today is to convey liveness timeout failures correctly with a precise error code - it's conveyed as a `CHIP_ERROR_TIMEOUT` when we should in fact, be sending something more descriptive. 

Levels above that are still WIP I believe..

mrjerryjohns, 2021-11-19:
@sagar-apple is this good enough for you?

mrjerryjohns, 2021-11-19:
@yunhanw-google to fix the emission of a more precise error code for liveness failure.

bzbarsky-apple, 2021-11-20:
@sagar-apple needs those "levels above" fixed is the point.

closed 2021-12-09 by andy31415