project-chip/connectedhomeip #11654: TC-GENDIAG-1.1 - General Diagnostics Cluster Attributes not able to be read or are not implemented

liamgonyea, 2021-11-10:
#### Problem
Some of the attributes on the General Diagnostics cluster are failing for different reasons

Failing Attributes: 
1.  total-operational-hours
Error: "CHIP:DMG: Error retrieving data from clusterId: 0x0000_0033, err = ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/platform/Linux/PlatformManagerImpl.cpp:328: CHIP Error 0x00000086: Valid time value is not available" 
2. network-interfaces
Problem: Lists the Hardware address, not the IPv4Address or IPv6Address. For "HardwareAddress: 6", does not list the full address but rather the number of byte segments in that field. The data not mentioned above is correct. 

Not-implemented Attributes:
3. active-hardware-faults
4. active-radio-faults
5. active-network-faults

#### Test Cases
TC-GENDIAG-1.1 | Attributes with server as DUT
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/general_diagnostics.adoc#tc-gendiag-1-1-attributes-with-server-as-dut

 #### Testing Setup
`chip-tool` and `chip-all-clusters-app`: commit `f51eeb39de1e58c8f55a13cfc59782e8e11c6948` (HEAD -> test_event_7, origin/test_event_7)

#### Logs
[GENDIAG Cluster Attributes Logs.txt](https://github.com/project-chip/connectedhomeip/files/7516635/GENDIAG.Cluster.Attributes.Logs.txt)

jmeg-sfy, 2021-11-18:
might have a some dupe here and there : 
https://github.com/project-chip/connectedhomeip/issues/11828
https://github.com/project-chip/connectedhomeip/issues/11899

liamgonyea, 2022-01-21:
### Update for TE 7.5: Issue Resolved
commit: `b0af6ba42b6ce494748f18ecf3e054a400b8f45b`

Each of the above issues were addressed for TE7.5. 

#### Lines of Interest from Logs
(Only from `consolidated_mac_accessory_TC-GENDIAG-1-1.txt`  as both logs are the same commands and same failures)
1.  total-operational-hours
> [1642731254261] [90350:3866645] CHIP: [TOO] GeneralDiagnostics.TotalOperationalHours response: 4
2. network-interfaces
> [1642731147501] [90316:3864797] CHIP: [TOO]     HardwareAddress: DCA632A849EA
3. active-hardware-faults
> [1642731325673] [90403:3868477] CHIP: [TOO] GeneralDiagnostics.ActiveHardwareFaults response: 0 entries
4. active-radio-faults
> [1642731361199] [90435:3869125] CHIP: [TOO] GeneralDiagnostics.ActiveRadioFaults response: 0 entries
5. active-network-faults
> [1642731397021] [90456:3871054] CHIP: [TOO] GeneralDiagnostics.ActiveNetworkFaults response: 0 entries

#### Logs
[consolidated_mac_accessory_TC-GENDIAG-1-1.txt](https://github.com/project-chip/connectedhomeip/files/7916330/consolidated_mac_accessory_TC-GENDIAG-1-1.txt)
[consolidated_mac_accessory_TC-GENDIAG-1-2.txt](https://github.com/project-chip/connectedhomeip/files/7916331/consolidated_mac_accessory_TC-GENDIAG-1-2.txt)

liamgonyea, 2022-01-21:
Closing issue, no longer seeing.

closed 2022-01-26 by raju-apple