project-chip/connectedhomeip #11670: "Null" check implementation in YAML framework

manjunathambiger23, 2021-11-11:
Summary:
As per spec the default value of some attributes are null and we need to check the default value as expected using Scripts but Null check is not implemented in YAML framework.

Spec Screenshot: 
[image]

Execution Screenshot:
[image]

bzbarsky-apple, 2022-01-07:
Null checking is implemented just fine.  But in the XML these attributes are not marked nullable.

manjunathambiger23, 2022-01-10:
@bzbarsky-apple we are facing same build error issue in Pump configuration cluster:
Spec screenshot:
[image]
Execution screenshot:
[image]

bzbarsky-apple, 2022-01-15:
> @bzbarsky-apple we are facing same build error issue in Pump configuration cluster:

I suggest filing separate issues per cluster that has this problem, but more importantly getting the domain owners for these clusters to fix those issues...

manjunathambiger23, 2022-01-19:
> > @bzbarsky-apple we are facing same build error issue in Pump configuration cluster:
> 
> I suggest filing separate issues per cluster that has this problem, but more importantly getting the domain owners for these clusters to fix those issues...

ok i will do it

manjunathambiger23, 2022-03-23:
https://github.com/project-chip/connectedhomeip/issues/16552

manjunathambiger23, 2022-03-23:
https://github.com/project-chip/connectedhomeip/issues/16550

manjunathambiger23, 2022-03-23:
https://github.com/project-chip/connectedhomeip/issues/16554

closed 2022-05-23 by bzbarsky-apple