project-chip/connectedhomeip #11677: [Pairing] How can we Setup pincode & Discriminator of CHIP end device?

maunishpatel1, 2021-11-11:
Hi,
I am trying to pair a CHIP device to the controller using chip-tool by using the command **"./chip-tool pairing ble-thread operationalDataset fabric-id setup-pin-code discriminator"**.

I want to connect any CHIP enabled device to the controller so my questions are:
**Are the Discriminator and pin code  common for all CHIP devices?**
if no, then **How can we get the Discriminator and pin code of the end device? can we get it using QR code or the BLE advertising?** 
And **can we pair the end device without using the discriminator and setup pin code?**

RudyGOGOGO, 2022-03-23:
Based on the doc (https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32), those parameters should be configurable
```
Discriminator: 3840 (configurable through menuconfig)
Setup-pin-code: 20202021 (configurable through menuconfig)
```
However, I don't find where I can configure them in the menuconfig, have you figured it out？
And one more question is: I could find the discriminator in the device log, however, I could not find the setupPinCode

closed 2022-09-27 by stale[bot]

isgmini, 2024-08-01:
Can someone answer the question? I have the same issue. I need to change the passcode and discriminator to pass the tests in the lab, keeping the default certificate.